ros-perception/image_pipeline #271: Release into ROS Lunar

130s, 2017-04-17:
[Some packages](http://repositories.ros.org/status_page/blocked_releases_lunar.html?s=6&r=1) want this package to be available.
Thanks!

mikaelarguedas, 2017-04-28:
@130s FYI: released in https://github.com/ros/rosdistro/pull/14581

@vrabaud This needs a new release including https://github.com/ros-perception/image_pipeline/commit/562332df73781c1e56ce9123542334cc1d91b143, without it all jobs on yakkety, zesty and stretch are failing. If you have any spare cycle in the next few days it will be great because thats one of the 3 packages preventing the release of ROS Lunar desktop_full

130s, 2017-04-29:
Thanks @mikaelarguedas

mikaelarguedas, 2017-04-29:
I'm going to reopen this given that a new release is needed to be able to release ROS Lunar

vrabaud, 2017-04-30:
And done !

closed 2017-04-30 by vrabaud