ros-perception/image_pipeline #272: Vertical misalignment of registered depth image compared to rectified RGB image

mindThomas, 2017-04-18:
This is a referenced issue from https://github.com/intel-ros/realsense/issues/233
Though the experienced problems are related to the registration process within the "depth_image_proc" code.

### System Configuration
| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | Yocto               |
| Kernel                | 4.4.3-yocto-standard                |
| ROS                   | indigo                    |
| ROS RealSense         | 1.7.2                |
| librealsense          | 1.12.1   (system lib)                |
| Camera Type-Firmware  | R200-1.0.71.06            |

### Expected Behavior
I would expect /camera/rgb/image_rect_color and /camera/depth_registered/sw_registered/image_rect_raw to be aligned.

### Actual Behavior
Please download the pictures below and try to match them (with transparancy).
[image]
[image]

Doing a transparent matching of the above test images results in the following horizontally mismatched picture.
[image]
Notice how the registered depth image has a 9 pixel tall black area in the top and a 20 pixel black bar in the right side. If the registered depth image is just moved 9 pixels upwards the color and depth image seem to match much better.
[image]

### Steps to Reproduce
Launch the rgbd nodelet.
`roslaunch realsense_camera r200_nodelet_rgbd.launch`
Capture images with rqt or similar

### Further Questions
Can someone elaborate on the difference between all the topics being published by the RGBD nodelet?
```
/camera/depth/camera_info
/camera/depth/disparity
/camera/depth/image
/camera/depth/image_raw
/camera/depth/image_rect
/camera/depth/image_rect_raw
/camera/depth/points
/camera/depth_registered/disparity
/camera/depth_registered/points
/camera/depth_registered/sw_registered/camera_info
/camera/depth_registered/sw_registered/image_rect
/camera/depth_registered/sw_registered/image_rect_raw
/camera/ir/camera_info
/camera/ir/image_raw
/camera/ir/image_rect_ir
/camera/ir2/camera_info
/camera/ir2/image_raw
/camera/rgb/camera_info
/camera/rgb/image_color
/camera/rgb/image_mono
/camera/rgb/image_raw
/camera/rgb/image_rect_color
/camera/rgb/image_rect_mono
```
What is the difference between raw and rectified in the depth image (there doesn't seem to be any difference, which could be explained from the fact that the depth image does not need distortion correction)?
What is the difference between raw and rectified RGB image? The only difference there seem to be is that the rectified RGB image is non-distorted. Is this correct? No conversion using intrinsics or extrinsics, eg. moving into a different frame?
What is the registered depth image? Is that simply each pixel of the depth image transformed into the optical RGB frame (image plane) and hence matching with the rectified RGB image?

k-okada, 2018-05-19:
I think this is the problem of extrinsic parameters provided from realsense driver, and (maybe) fixed by https://github.com/intel-ros/realsense/issues/233#issuecomment-361920190

SteveMacenski, 2019-11-03:
Intel issue fixed, closing

closed 2019-11-03 by SteveMacenski