ros-perception/image_pipeline #275: stereo_calibration doesn't start

ashnarayan13, 2017-05-22:
I am able to calibrate the cameras individually. But when I run the stereo_calibration using the http://wiki.ros.org/camera_calibration/Tutorials/StereoCalibration I the calibration windows opens and turns grey.
I increased the buffer for the cameras to 1GB and tested if the cameras were working in stereo, I get images when view the topics on rviz. 
Any suggestions?

trigal, 2017-05-22:
do you have synchronized camera messages? if they're not, use the approximate flag, like --approximate=0.1  and let us know!

ashnarayan13, 2017-05-22:
That worked! Thanks. 
When I try to commit the calibration, I get the 
QObject::~QObject: Timers cannot be stopped from another thread
Is this a camera problem or the calibration window problem?

trigal, 2017-05-22:
I do not have an answer for this issue but I remember that my colleagues also had problems with that. Maybe you need a suitable driver to "accept" the incoming calibration, don't kwnw. What we usually do is to save the calibration (in /tmp) and than we use them in our ad-hoc camera driver, but I'm sure the other guys here can say something better ;-)

ashnarayan13, 2017-11-14:
This seems to be occuring at random. I could get it to commit the config and it fails to do it sometimes. 
I worked around it by saving the calibration and then reading the calibration data when I start my ros node.

closed 2017-11-14 by ashnarayan13

WwYyFan, 2018-08-22:
@ashnarayan13 hi,
I also encountered this problem. How did you solve it?

efun@efun:~/catkin_ws$ rosrun camera_calibration cameracalibrator.py --size 8x6 --square 0.108 right:=/mynteye/right/image_raw left:=/mynteye/left/image_raw right_camera:=/mynteye/right left_camera:=/mynteye/left
('Waiting for service', '/mynteye/left/set_camera_info', '...')
^CService not found
('Waiting for service', '/mynteye/right/set_camera_info', '...')
Service not found