ros-perception/image_pipeline #281: calibration parameters varying across multiple attempts for monocular camera

arpit15, 2017-06-29:
I am trying to [camera calibration tool](http://wiki.ros.org/camera_calibration) for calibrating the rgb sensor in realsense sr300. However I get hugely varying values of the camera intrinsics across different attempts.
Could anyone suggests what might be wrong with the method or my approach to it? This seems really weird. For calibration I place the pattern on a table and move the camera around. 

The output of the node for some attempts.

camera matrix
623.431503 0.000000 333.741849
0.000000 625.430581 241.285772
0.000000 0.000000 1.000000

distortion
0.128823 -0.340227 0.002485 0.010646 0.000000

rectification
1.000000 0.000000 0.000000
0.000000 1.000000 0.000000
0.000000 0.000000 1.000000

projection
627.600403 0.000000 338.896375 0.000000
0.000000 635.116821 241.670688 0.000000
0.000000 0.000000 1.000000 0.000000

camera matrix
639.381239 0.000000 322.010453
0.000000 634.598648 238.819443
0.000000 0.000000 1.000000

distortion
0.196652 -0.437027 0.001988 0.001229 0.000000

rectification
1.000000 0.000000 0.000000
0.000000 1.000000 0.000000
0.000000 0.000000 1.000000

projection
652.564636 0.000000 322.103325 0.000000
0.000000 647.487488 238.939412 0.000000
0.000000 0.000000 1.000000 0.000000

camera matrix
625.023217 0.000000 327.590344
0.000000 628.041437 254.452492
0.000000 0.000000 1.000000

distortion
0.185817 -0.384493 0.008467 0.009156 0.000000

rectification
1.000000 0.000000 0.000000
0.000000 1.000000 0.000000
0.000000 0.000000 1.000000

projection
639.086548 0.000000 331.864149 0.000000
0.000000 643.823853 256.924545 0.000000
0.000000 0.000000 1.000000 0.000000

camera matrix
654.067639 0.000000 327.137576
0.000000 656.213262 250.301701
0.000000 0.000000 1.000000

distortion
0.272707 -0.619009 0.002485 0.004020 0.000000

rectification
1.000000 0.000000 0.000000
0.000000 1.000000 0.000000
0.000000 0.000000 1.000000

projection
672.697571 0.000000 328.531188 0.000000
0.000000 675.860718 250.666310 0.000000
0.000000 0.000000 1.000000 0.000000

SubMishMar, 2017-07-06:
Even I face the same issue! I am using it for intel realsense euclid's RGB camera. 
Just a question, how do you know if the calibration is good or not? Is there something like calculating the reprojection errors? 

After you are done calibrating you will see "lin" written on top right corner with a black fine straight line across the screen. What does it mean?

nalt, 2018-04-05:
I noticed similar problems with Logitech cameras. Did you fix it?

JWhitleyAStuff, 2019-04-25:
@arpit15 / @SubMishMar / @nalt - Could you possibly try the fix in #356 and see if your issue is resolved?

SteveMacenski, 2019-04-26:
@all this is expected and you should always check against problems like this in calibration. Minimizing energy doesn't always mean that accuracy ~ percision. 

When calibrating cameras with even other methods, I've come to expect the results could vary by *many* pixels between attempts. You should always error check after a calibration to make sure your reprojection error is aligned with that and within whatever criteria your design requires.

closed 2019-04-26 by SteveMacenski