ros-perception/image_pipeline #283: Critical error when using image_view nodelet

gui2dev, 2017-07-11:
Hi !

I'm using kinetic on ubuntu 16.04 on an Intel Joule 570x.
I'm trying to use image_view/image nodelet and get the following error :
`[ INFO] [1499780848.028943396]: Using transport "raw"
init done

(nodelet:19721): GLib-GObject-WARNING **: invalid uninstantiatable type '(null)' in cast to 'GtkWidget'

(nodelet:19721): GLib-GObject-WARNING **: instance of invalid non-instantiatable type '(null)'

(nodelet:19721): GLib-GObject-CRITICAL **: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
constructor
`

I'm using a launch file to start the nodelet :

`
<arg name="nodelet_manager_name" value="nodelet_manager" />

<node pkg="nodelet" type="nodelet" name="$(arg nodelet_manager_name)" args="manager" output="screen"/>

<node pkg="nodelet" type="nodelet" name="image_view_nodelet" args="load image_view/image $(arg nodelet_manager_name)">
    <remap from="/image" to="camera/image_raw"/>
    <param name="image_transport" type="string" value="raw"/> 

  <node pkg="nodelet" type="nodelet" name="raytrix_cam_nodelet" args="load cam/cam_nodelet $(arg nodelet_manager_name)">
</node>`

I wrote and compiled the cam/cam_nodelet myself.

Thank you for your support !

k-okada, 2017-07-11:
if this is a same situation as I rporeted at
https://github.com/ros-perception/image_pipeline/pull/279,
please try something like
https://github.com/ros-perception/opencv_apps/pull/67 or run nodelet with 1
thread,

gui2dev, 2017-07-11:
When running the nodelet as in #279, it freezes after printing the same GLib Warning / errors.

I tried to run the nodelet with one thread, setting num_worker_threads param to 1 in my launch file and it raised the same issue.

Should I try #279 workaround and recompile image_pipeline with the fix ?

k-okada, 2017-07-12:
> Should I try #279
<https://github.com/ros-perception/image_pipeline/pull/279> workaround and
recompile image_pipeline with the fix ?

Please try it

gui2dev, 2017-07-18:
I just did recompile opencv_apps and image_pipeline from source.
Same issue raised.
Any clue what should I investigate first ?

k-okada, 2017-07-20:
that's strange  https://github.com/ros-perception/image_pipeline/pull/279/files works for me
Did you merge #279 patch before you recompile it?
```
git remote add k-okada http://github.com/k-okada/image_pipeline
git fetch --all
git merge k-okada/fix_freeze 
```

k-okada, 2018-05-19:
see more discussion in #331 and proposed fix at #337

JWhitleyAStuff, 2019-04-25:
Assumed fixed by #337.

closed 2019-04-25 by JWhitleyAStuff