ros-perception/image_pipeline #285: tarfile_calibration does not visualize rectified images after calibration

trigal, 2017-07-18:
in tarfile_calibration.py there is a waitKey define with 6 ms as delay (as reported in [opencv documentation](http://docs.opencv.org/2.4/modules/highgui/doc/user_interface.html#waitkey) . 
This doesn't make sense since waitkey() should wait for a pressed key, then releasing the while inside the display() define. 

```python
def waitkey():
    k = cv2.waitKey(6)
    return k

def display(win_name, img):
    cv2.namedWindow(win_name, cv2.WINDOW_NORMAL)
    cv2.imshow( win_name,  numpy.asarray( img[:,:] ))
    k=-1
    while k ==-1:
        k=waitkey()
    cv2.destroyWindow(win_name)
    if k in [27, ord('q')]:
        rospy.signal_shutdown('Quit')
```

this should be the desired behavior: 
```python
def waitkey():
    k = cv2.waitKey(0)
    return k
```

k-okada, 2018-05-19:
I have rewrite #286  and post new PR #340, but not sure how this related to the "does not visualize rectified images after calibration" can you provide sample tar file and instructions to how to reproduce this problem?

JWhitleyAStuff, 2019-04-18:
Closing stale PR which appears to also be fixed by #340. Can re-open if OP replies or offers feedback.

PR #340 (2019-04-19): use waitKey(0) instead of while loop
rewrited version of #286

Closes #286
Closes #285
closed 2019-04-18 by JWhitleyAStuff