ros-perception/image_pipeline #287: import cv2 ImportError: libvpx.so.3: cannot open shared object file: No such file or directory

doisyg, 2017-07-19:
Hello,
I have a dependency error on kinetic and ubuntu 16.04.
Any idea where it is coming from? I did not install opencv from source.

epiception, 2017-07-19:
Did you try running `sudo apt-get install libvpx3` in a new terminal window?

doisyg, 2017-07-19:
Yes:
```
$ sudo apt-get install libvpx3
[sudo] password for gd: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libvpx3 is already the newest version (1.6.0-561-g065ba0c-1).
libvpx3 set to manually installed.
```

vrabaud, 2017-07-19:
You must have libvpx from somewhere else (maybe a ppa).
These are the official version:
https://packages.ubuntu.com/xenial/arm64/libvpx3/filelist
Your solution is to recompile OpenCV (you can include it in your catkin workspace).

doisyg, 2017-07-19:
Thanks a lot!
It was indeed fetching the package from a ppa that I completly forgot to remove from my source list

closed 2017-07-19 by doisyg