ros-perception/image_pipeline #288: image_saver node doesn't work with usb_cam package due to unexpected distortion_model

danielsnider, 2017-08-05:
I'm trying to use `image_saver` to save images from my usb webcam stream from the `usb_cam` [package](wiki.ros.org/usb_cam) but I get an error and it doesn't save any images.

Here's what I run:
```
<launch>
  <node name="usb_cam1" pkg="usb_cam" type="usb_cam_node" output="screen" respawn="true">
      <param name="camera_frame_id" value="usb_cam1"/>
      <param name="camera_name" value="usb_cam1"/>
      <param name="video_device" value="/dev/video0"/>
      <param name="image_width" value="640"/>
      <param name="image_height" value="480"/>
      <param name="framerate" value="30"/>
      <param name="pixel_format" value="yuyv"/>
      <param name="io_method" value="mmap"/>
  </node>

  <node pkg="image_view" type="image_saver" name="image_saver" output="screen" clear_params="true">
    <param name="filename_format" value="%04i.png"/>
    <param name="save_all_image" value="true"/>
    <remap from="image" to="usb_cam1/image_raw" />
  </node>
</launch>
```
But I get this error:
```
[ INFO][7f25ad29c780][/image_saver/saveImage:168]: Saved image 0072.png
[ERROR][7f25ad29c780][/image_saver/writeCalibrationIni:91]: Videre INI format can only save calibrations using the plumb bob distortion model. Use the YAML format inste
ad.
        distortion_model = '', expected 'plumb_bob'
        D.size() = 0, expected 5
```

Can't image_saver just save images and not worry about the distortion model if it's not set?

Thanks!

k-okada, 2017-08-06:
I think best way is to fix source code, but as a workaroud, publish dummy camera info at usb_cam side
```
      <param name="camera_info_url" value="file://$(env HOME)/.ros/camera_info/camera.yaml" />
```
or escape being subscribe camera_info image_saver side
```
  <node pkg="image_view" type="image_saver" name="image_saver"
        output="screen" clear_params="true">
    <param name="filename_format" value="%04i.png"/>
    <param name="save_all_image" value="true"/>
    <remap from="image" to="usb_cam1/image_raw" />
    <remap from="usb_cam1/camera_info" to="usb_cam1/camera_info_dummy" />
  </node>
```

mikeferguson, 2024-01-17:
This seems like a downstream configuration issue - not something we should change here for an invalid configuration

closed 2024-01-17 by mikeferguson