ros-perception/image_pipeline #289: Allow for image saving after pausing bag file

Chambana, 2017-08-09:
When my bag file is playing, I can easily click the "Save" icon and write the current frame from my streaming camera feed to disk.  However, when I pause my bag file playback to grab a specific image, the "Save" icon no longer functions and I can't save the paused image being displayed in the window.  Unless I'm doing something wrong, this is a feature request.  Thanks. 

How I'm executing the node:
      rosrun image_view image_view image:=/usb_cam/image_raw compressed _autosize:=true

Using ROS Kinetic on Ubuntu 16.04.3 LTS

SteveMacenski, 2019-11-04:
PRs welcome

mikeferguson, 2024-01-19:
Kinetic is EOL, not sure if this is an issue on more recent versions

closed 2024-01-19 by mikeferguson