ros-perception/image_pipeline #290: Disparity Map shown, but no pointcloud msg is published using stereo_image_proc

jomocl, 2017-08-22:
I am getting a disparity Image and rectified images from the stereo_image_proc it is launched using:

<node name="stereo_image_proc" pkg="stereo_image_proc" type="stereo_image_proc" output="screen" ns="claas_camera_node" >
</node>
<node name="stereo_cam" pkg="image_view" type="stereo_view" output="screen" >
<remap from="stereo" to="/claas_camera_node"/>
<remap from="image" to="image_rect_color" />
</node>	

but looking in rviz there is nothing published on the pointcloud message?

paulsammut, 2017-12-12:
Load up rqt_graph and make sure that stereo_image_proc node looks good with all necassary messages going to it. 

Then, run make sure that the point cloud topic is being published. Run this command:

rosparam echo /name/of/point/cloud/topic 

A possible issue is that the frame associate with your laser scans isn't properly setup with tf. Good luck!

Syed-Izzat-Ullah, 2019-04-17:
> I am getting a disparity Image and rectified images from the stereo_image_proc it is launched using:
> 
> but looking in rviz there is nothing published on the pointcloud message?

Hello @paulsammut  have you been able to solve this problem? I am also getting disparity images but unable to display pointcloud. my poincloud topic only returns consistent same values and in rviz I can only see a small dot. please let me know if you know what mistakes I made. regards

SteveMacenski, 2019-04-17:
Closing this issue from inactivity (and probably resolved after 2 years).

closed 2019-04-17 by SteveMacenski

nick-paul, 2019-12-18:
I am having a similar issue with the current version of `stereo_image_proc` A point cloud is published with the correct transform but only a single point is visible at the origin of the transform in rviz.

These unanswered questions on answers.ros.org are describing the same issue and go into a bit more detail:

  - https://answers.ros.org/question/333144/stereo_image_proc-does-not-generate-pointcloud/
  - https://answers.ros.org/question/321282/disparity-image-is-displayed-but-pointcloud-shows-only-a-small-dot-in-rviz/

This bag contains raw camera messages and the generated point clouds:
[points_and_cams.bag.zip](https://github.com/ros-perception/image_pipeline/files/3979506/points_and_cams.bag.zip)

Any idea what the problem might be?
Thanks

leander-dsouza, 2021-06-29:
### Gazebo Setup

[image]

* I faced the exact same issue while simulating a custom stereo camera in Gazebo.<br/> This happens if you don't set the **`hackBaseline`** parameter.

* Following is the Gazebo plugin I had used:

		

		<gazebo reference="left_stereo_camera">

			<sensor type="multicamera" name="stereocamera">
				<always_on>true</always_on>
				<update_rate>10</update_rate>
				<camera name="left">
					<pose>0 0 0 0 0 0</pose>
					<horizontal_fov>${radians(front_camera_horizontal_fov)}</horizontal_fov>
					<image>
						<width>640</width>
						<height>480</height>
						<format>RGB_INT8</format>
					</image>
					<clip>
						<near>0.1</near>
						<far>100</far>
					</clip>
				</camera>
				<camera name="right">
					<pose>0 -${front_camera_baseline} 0 0 0 0</pose>
					<horizontal_fov>${radians(front_camera_horizontal_fov)}</horizontal_fov>
					<image>
						<width>640</width>
						<height>480</height>
						<format>RGB_INT8</format>
					</image>
					<clip>
						<near>0.1</near>
						<far>100</far>
					</clip>
				</camera>        
				<plugin name="stereo_camera_controller" filename="libgazebo_ros_multicamera.so">
					<robotNamespace>/</robotNamespace>
					<cameraName>stereocamera</cameraName>
					<alwaysOn>true</alwaysOn>
					<updateRate>10</updateRate>
					<imageTopicName>image_raw</imageTopicName>
					<cameraInfoTopicName>camera_info</cameraInfoTopicName>
					<frameName>stereo_optical</frameName>
					<hackBaseline>${front_camera_baseline}</hackBaseline>
					<distortion_k1>0.0</distortion_k1>
					<distortion_k2>0.0</distortion_k2>
					<distortion_k3>0.0</distortion_k3>
					<distortion_t1>0.0</distortion_t1>
					<distortion_t2>0.0</distortion_t2>
				</plugin>
			</sensor>
		</gazebo>

* If **`hackBaseline`** is set to `0.00`:<br/><br/> Here you can see the dot in RViz at the centre of the optical frame.
[image]

* If **`hackBaseline`** is set to `0.06`:
[image]

JMadhup, 2022-06-17:
@leander-dsouza my hackBaseLine is already set to 0.07, I have tried other values as well. I am getting the disparity image but still cannot visualize the point cloud in rviz

leander-dsouza, 2022-06-19:
This might seem trivial, but are you using ROS2? as the `hackBaseline` parameter changes its alias to `hack_baseline`.

JMadhup, 2022-06-20:
I am using ROS noetic