ros-perception/image_pipeline #291: C++ 11 is necessary to compile with latest image_geometry package

alexzzhu, 2017-08-29:
The image_geometry package now compiles with C++ 11 by default: (https://github.com/ros-perception/vision_opencv/blob/8b5bbcbc1ce65734dc600695487909e0c67c1033/image_geometry/CMakeLists.txt#L15) , and uses std::shared_ptr: (https://github.com/ros-perception/vision_opencv/blob/8b5bbcbc1ce65734dc600695487909e0c67c1033/image_geometry/include/image_geometry/pinhole_camera_model.h#L273). 

At present, image_pipeline does not compile with C++ 11, which causes the compiler to crash when compiling with image_geometry.

Chambana, 2017-09-03:
I believe I'm having the same issue. The build fails with:

[ 52%] Building CXX object image_pipeline/image_proc/CMakeFiles/image_proc.dir/src/nodelets/rectify.cpp.o
In file included from /home/drew/catkin_ws/src/image_pipeline/image_proc/include/image_proc/processor.h:38:0,
                 from /home/drew/catkin_ws/src/image_pipeline/image_proc/src/libimage_proc/processor.cpp:34:
/home/drew/catkin_ws/src/vision_opencv/image_geometry/include/image_geometry/pinhole_camera_model.h:273:8: error: ‘shared_ptr’ in namespace ‘std’ does not name a template type
   std::shared_ptr<Cache> cache_; // Holds cached data for internal use

alexzzhu, 2017-09-04:
This is resolved by specifying c++ 11 in the CMakeLists:
SET(CMAKE_CXX_FLAGS -std=c++11)

If having c++ 11 is an acceptable assumption moving forwards, I can make a PR to put this into the main branch.

Chambana, 2017-09-04:
Yes, please do

alexzzhu, 2017-09-08:
PR developed: https://github.com/ros-perception/image_pipeline/pull/292

vrabaud, 2017-09-18:
Sorry for the delay, 98670bdfc9fbd22a3ee7f4e75c3ac66965fcbb5d fixes it properly.

closed 2017-09-18 by vrabaud