ros-perception/image_pipeline #296: non authorised data type

MalcolmMielle, 2017-09-21:
Hej,

Against the latest version of opencv3, `processor.h` uses non authorised data type for the Mat, line180

```
  mutable cv::Mat_<uint32_t> wavefront_;
  mutable cv::Mat_<uint8_t> region_types_;
  // scratch buffer for dense point cloud
```

I have this error on compile
```
/usr/local/include/opencv2/core/mat.hpp:2150:50: error: no type named ‘channel_type’ in ‘class cv::DataType<unsigned int>’
     typedef typename DataType<_Tp>::channel_type channel_type;

```

See [here](https://github.com/opencv/opencv/issues/9677). I'm not exactly sure which type I should replace them by

MalcolmMielle, 2017-09-21:
I replace the three offending lines by 
```
  
  mutable cv::Mat_<int> labels_;
  mutable cv::Mat_<int> wavefront_;
  mutable cv::Mat_<char> region_types_;
  // scratch buffer for dense point cloud
```

If it is indeed the good type fix, let me know and I'll push the changes in a merge request

SteveMacenski, 2019-11-03:
PR welcome

mikeferguson, 2024-01-20:
Jammy is opencv 4.5 - so I presume we've fixed these issues at this point

closed 2024-01-20 by mikeferguson