ros-perception/image_pipeline #308: Problem compiling with latest traits.hpp

FallenBuckeye, 2017-11-19:
With latest Opencv2, Opencv_contrib, Ceres with eigen3, etc, using the sfm scene reconstruction example cooied into eclipse, I receive the following compile error:
/home/bongo/opencv/modules/core/include/opencv2/core/traits.hpp: In instantiation of ‘struct cv::traits::Type<std::__cxx11::basic_string<char> >’:
/home/bongo/opencv/modules/core/include/opencv2/core/mat.inl.hpp:85:32:   required from ‘cv::_InputArray::_InputArray(const std::vector<_Tp>&) [with _Tp = std::__cxx11::basic_string<char>]’
../src/recon.cpp:61:81:   required from here
/home/bongo/opencv/modules/core/include/opencv2/core/traits.hpp:366:8: error: ‘type’ is not a member of ‘cv::DataType<std::__cxx11::basic_string<char> >’
 { enum { value = DataType<T>::type }; };
 This is a fresh build and git from Nov 19, 2017. Can anyone help?

vrabaud, 2017-11-20:
Where is the code you are trying to compile ?

FallenBuckeye, 2017-11-24:
Thanks for your response. My source code was the example code the
opencv_contrib sfm scene recon example. The error seemed to indicate that
the sfm was using a deprecated member of traits, so I suppose I should have
posted my problem to contrib. I checked the last check-in on sfm and it
seemed to match opencv 3.2.0 release, so I did a git on that version and
the problem went away. Should I post this somewhere? Accept my apologies
for poor decorum.

llvs, 2018-02-13:
To me, the following workaround worked in Windows/Visual Studio:
`#define OPENCV_TRAITS_ENABLE_DEPRECATED`
Using current OpenCV git version 3.4.0+

whuhit, 2019-01-30:
> To me, the following workaround worked in Windows/Visual Studio:
> `#define OPENCV_TRAITS_ENABLE_DEPRECATED`
> Using current OpenCV git version 3.4.0+

it works

JWhitleyAStuff, 2019-04-25:
Closing due to resolution.

closed 2019-04-25 by JWhitleyAStuff

nidhiratanpara, 2020-03-18:
This does not work with the workaround yet. I am using the latest opencv version.

nidhiratanpara, 2020-03-18:
@JWhitleyWork  This does not work with the given workaround yet. I am using the latest opencv version.

JWhitleyWork, 2020-03-19:
@nidhiratanpara This is a very old issue. Please create a new issue with your environment details and the error you're seeing.