ros-perception/image_pipeline #310: OpenCV3 Regression error

miquelmassot, 2017-11-30:
See https://github.com/opencv/opencv/issues/7599

In https://github.com/ros-perception/image_pipeline/blob/2589ed28251268a9657f4cc037373b063cca9958/stereo_image_proc/include/stereo_image_proc/processor.h#L180 and https://github.com/ros-perception/image_pipeline/blob/2589ed28251268a9657f4cc037373b063cca9958/stereo_image_proc/include/stereo_image_proc/processor.h#L181 two cv::Mat_<uint32_t> matrices are declared but not used, and their types are no longer supported by OpenCV3.

Related from https://github.com/srv/srv_tools/issues/17 and https://github.com/ros-infrastructure/rep/pull/140

tspicer01, 2018-03-06:
Is there an ETA on when we could expect the fix ?
We would really like to move to the new version of opencv3.

JWhitleyAStuff, 2019-04-25:
Closed by #315.

closed 2019-04-25 by JWhitleyAStuff