ros-perception/image_pipeline #312: stereo camera disparity map looks well,but point cloud looks strange in rviz.

shmm91, 2017-12-11:
Hello, everyone! 
I have done the stereo camera disparity, and i want to get the point cloud.
Then, I use the code in image_pipeline package(stereo_image_proc -> src -> nodelets -> point_cloud2.cpp), and rewrite it.
But I get this result, the pointcloud2 looks strange, could someone help me?! Thank you!
[image]

paulsammut, 2017-12-12:
Hi! Could you give more info on your setup? What camera do you have? Have you calibrated it? Seee here if not: http://wiki.ros.org/camera_calibration.

Also, the disparity image actually doesn't look too bad. It also looks like your test area isn't very well lit, and the image doesn't look that good. You are using indoor florescent lighting which flickers and can have bad effects on the stereo image. I recommend definitely calibrating the camera with the checkerboard (make your own) and fix the exposure on the camera if you can. Good luck!

shmm91, 2017-12-13:
Thank you! I have made stereo camera(matrix vision bluefox2 camera) calibration using ros package. This is one of the camera calibration file.

image_width: 752
image_height: 480
camera_name: mv_26806343
camera_matrix:
  rows: 3
  cols: 3
  data: [483.0500651879798, 0, 383.7668152619602, 0, 485.226588991044, 228.5179979149652, 0, 0, 1]
distortion_model: plumb_bob
distortion_coefficients:
  rows: 1
  cols: 5
  data: [-0.3321920298358194, 0.1047044621854059, 0.003132906729067026, -0.002535682618479629, 0]
rectification_matrix:
  rows: 3
  cols: 3
  data: [0.9998591468441322, 0.01204174307000691, 0.01169114604989564, -0.01203101395290746, 0.9999271370075643, -0.0009876128396240638, -0.01170218677807683, 0.000846817389987077, 0.9999311684935714]
projection_matrix:
  rows: 3
  cols: 4
  data: [458.7297328525751, 0, 365.8258590698242, 0, 0, 458.7297328525751, 236.6724758148193, 0, 0, 0, 1, 0]

The point cloud processing code.
void image_pcl(const sensor_msgs::ImageConstPtr& l_image_msg, const sensor_msgs::CameraInfoConstPtr& l_info_msg, const sensor_msgs::CameraInfoConstPtr& r_info_msg, const sensor_msgs::ImageConstPtr& disp_msg)
{
	cv::Mat_<cv::Vec3f> points_mat;
	image_geometry::StereoCameraModel stere_model;
	stere_model.fromCameraInfo(l_info_msg, r_info_msg);

	cv::Mat disp_raw = cv_bridge::toCvShare(disp_msg, "mono8")->image;
	cv::Mat_<int16_t> disparity16;
	disp_raw.convertTo(disparity16,disparity16.type());

	const cv::Mat_<uint8_t> l_image = cv_bridge::toCvShare(l_image_msg,sensor_msgs::image_encodings::MONO8)->image;

	DisparityImagePtr disp = boost::make_shared<DisparityImage>();
	disp->header = l_info_msg->header;
	disp->image.header = l_info_msg->header;
	disp_proc(l_image,stere_model,disparity16,*disp);

	double cx_l = stere_model.left().cx();
	double cx_r = stere_model.right().cx();
	if(cx_l != cx_r)
	{
		cv::Mat_<float> disp_image(disp->image.height,disp->image.width,reinterpret_cast<float*>(&disp->image.data[0]),disp->image.step);
		cv::subtract(disp_image,cv::Scalar(cx_l - cx_r),disp_image);
	}

	// calculate point cloud
	const sensor_msgs::Image& dimage_ = disp->image;
	const cv::Mat_<float> dmat_(dimage_.height,dimage_.width,(float*)&dimage_.data[0],dimage_.step);
	stere_model.projectDisparityImageTo3d(dmat_,points_mat,true);
	cv::Mat_<cv::Vec3f> mat = points_mat;

	PointCloud2Ptr points_msg = boost::make_shared<PointCloud2>();
	points_msg->header = disp->header;
	points_msg->header.frame_id = "stereo_frame";
	points_msg->header.stamp = ros::Time::now();
	points_msg->height = mat.rows;
	points_msg->width = mat.cols;
	points_msg->is_dense = true;

	sensor_msgs::PointCloud2Modifier pcd_modifier(*points_msg);
	pcd_modifier.setPointCloud2FieldsByString(2,"xyz","rgb");

	sensor_msgs::PointCloud2Iterator<float> iter_x(*points_msg,"x");
	sensor_msgs::PointCloud2Iterator<float> iter_y(*points_msg,"y");
	sensor_msgs::PointCloud2Iterator<float> iter_z(*points_msg,"z");

	sensor_msgs::PointCloud2Iterator<uint8_t> iter_r(*points_msg,"r");
	sensor_msgs::PointCloud2Iterator<uint8_t> iter_g(*points_msg,"g");
	sensor_msgs::PointCloud2Iterator<uint8_t> iter_b(*points_msg,"b");

	float bad_point = std::numeric_limits<float>::quiet_NaN();
	for(int v = 0; v < mat.rows; ++v)
	{
		for(int u = 0; u < mat.cols; ++u, ++iter_x, ++iter_y, ++iter_z)
		{
			if(isValidPoint(mat(v,u)))
			{
				*iter_x = mat(v,u)[0];
				*iter_y = mat(v,u)[1];
				*iter_z = mat(v,u)[2];
			}
			else
			{
				*iter_x = *iter_y = *iter_z = bad_point;
			}
		}
	}
    // Fill color
	namespace enc = sensor_msgs::image_encodings;
	cv_bridge::CvImagePtr cv_ptr;
	cv_ptr = cv_bridge::toCvCopy(l_image_msg, sensor_msgs::image_encodings::BGR8);
	const std::string& encoding = cv_ptr->encoding;
	if(encoding == enc::BGR8)
	{
		const cv::Mat_<cv::Vec3b> color(cv_ptr->image.rows,cv_ptr->image.cols,(cv::Vec3b*)&cv_ptr->image.data[0],cv_ptr->image.step);
		for(int v = 0; v < mat.rows; ++v)
		{
			for(int u = 0; u < mat.cols; ++u, ++iter_r, ++iter_g, ++iter_b)
			{
				const cv::Vec3b& bgr = color(v,u);
				*iter_r = bgr[0];
				*iter_g = bgr[1];
				*iter_b = bgr[2];
			}
		}
	}
	pub_points.publish(points_msg);
}

syn1993, 2018-04-07:
Hi, have you solved the problem? I got the same problem.

k-okada, 2018-04-08:
can someone provide bag file reproducing this error?

JWhitleyAStuff, 2019-04-25:
Closing due to lack of response.

closed 2019-04-25 by JWhitleyAStuff