ros-perception/image_pipeline #325: Camera calibration should ignore frames with large motion (mainly for rolling shutter)

nalt, 2018-04-05:
Camera calibration takes images out of the video stream while the camera is moving. This means that slightly blurred images are used for calibration, which reduces the location accuracy of the corners points. Even worse, for rolling shutter cameras, the locations of corner points get shifted. This shift easily amounts to several pixels between the upper and the lower part of the pattern.

I noticed that calibration results differ a lot between several runs with rolling shutter cameras. A quick test revealed that this changes as soon as frames with large motion are rejected.

Did anyone see similar problems?

jack-digilabs, 2019-01-04:
I had a horrible problem with my "calibrated camera" being so off, that in RVIZ, it appeared to be pitching down 15 degrees in RVIZ. This caused issues like poor aruco marker detections who's axis sputtered and flipped around as well as detections that should have been straight ahead of the robot appearing 1 meter underground.

My Solution: Cover calibration board with hand until the board is steady then remove hand until a image was taken (prints out in terminal). Then recover board with hand and move to the next spot.

Life gets much easier when the cameras are calibrated properly. Possibly helpful to #356 and #281.

JWhitleyAStuff, 2019-04-25:
@nalt - I know this is an old issue but do you have any suggestions on how we might implement the rejection using an algorithm? (or, better yet, a PR implementing it)

PR #356 (2019-04-26): Enhance calibration of rolling shutter cameras (fixes #325, possibly #281)
This pull request aims to improve the calibration accuracy of rolling shutter cameras by avoiding picking frames where the chessboard pattern is moving in the image frame. This should fix #325 .
This is controlled by the --max-chessboard-speed option that allows setting a maximum allowed speed in pixels/frame for the chessboard. If the chessboard is moving faster than the allowed speed, the frame is not picked to be used for calibration.

The chessboard speed is determined by measuring the average distance that each che [... cut]
commit b20c5ca5 (2019-04-26): Merge pull request #356 from sevangelatos/feature/calibrator_rolling_shutter

Enhance calibration of rolling shutter cameras (fixes #325, possibly #281)
PR #415 (2019-05-05): Enhance calibration of rolling shutter cameras backport melodic->kinetic
This pull request aims to improve the calibration accuracy of rolling shutter cameras by avoiding picking frames where the chessboard pattern is moving in the image frame. This should fix #325 .
This is controlled by the --max-chessboard-speed option that allows setting a maximum allowed speed in pixels/frame for the chessboard. If the chessboard is moving faster than the allowed speed, the frame is not picked to be used for calibration.

The chessboard speed is determined by measuring the average distance that each chess [... cut]
closed 2019-04-26 by JWhitleyAStuff