ros-perception/image_pipeline #330: filename_format parameter cannot by given convention

tik0, 2018-05-07:
I use the image_saver node to store images.
Unfortunately, the syntax described in the [wiki](http://wiki.ros.org/image_view#image_view.2BAC8-diamondback.image_saver) does not work, due to parameter server issues.
When I execute `rosrun image_view image_saver filename_format:=cam_%04d.png`, the parameter server says:

```
terminate called after throwing an instance of 'ros::InvalidNameException'
  what():  Character [%] at element [4] is not valid in Graph Resource Name [cam_%04d.png].  Valid characters are a-z, A-Z, 0-9, / and _.
Aborted (core dumped)
```

k-okada, 2018-05-08:
hi, that's parameter, so you should type
```
rosrun image_view image_saver _filename_format:=cam_%04d.png
```

tik0, 2018-05-08:
You're right, thanks for the reply!

closed 2018-05-08 by tik0