ros-perception/image_pipeline #331: [image_view] window hangs because callback can be dispatched from multiple threads

yoshito-okada, 2018-05-08:
I found that the image_view nodelet (image_view/image) does not update its window and hangs in some cases.  I think this is because the nodelet manager can execute a callback in multiple threads even if the callback is associated to a threaded callback queue (see https://github.com/ros/nodelet_core/blob/2f25b775fd0241fc949a5fc002e718dbd142d5d7/nodelet/src/callback_queue_manager.cpp#L206), although the callback must be called by a specific thread to successfully update the window.

The following is a launch script to reproduce the problem.  I am afraid that the problem may not be reproduced in different environment because the problem depends on environment-specific threading but I hope this is helpful.
```
<launch>
    <node name="image_view_node" pkg="nodelet" type="nodelet" 
          args="manager image_view_node" output="screen"/>

    <node name="image_view_loader" pkg="nodelet" type="nodelet" 
          args="load image_view/image image_view_node" output="screen"/>

    <node name="image_view_loader2" pkg="nodelet" type="nodelet" 
          args="load image_view/image image_view_node" output="screen"/>

    <!-- An external node publishing sensor_msgs/Image on the "image" topic is required -->
</launch>
```

I think the only solution is to add a new thread updating the window and am writing a workaround.  Please see if interested.
https://github.com/yoshito-okada/image_pipeline/tree/fix_image_view_nodelet

yoshito-okada, 2018-05-10:
Sorry, I put a wrong launch script in my original post.  The following one would reproduce the problem.

```
<launch>
    <node name="image_view_node" pkg="nodelet" type="nodelet" 
          args="manager image_view_node" output="screen"/>

    <node name="image_view_loader" pkg="nodelet" type="nodelet" 
          args="load image_view/image image_view_node" output="screen">
        <remap from="image" to="image_mono"/>
    </node>

    <node name="image_proc_loader" pkg="nodelet" type="nodelet" 
          args="load image_proc/debayer image_view_node" output="screen">
        <remap from="image_raw" to="image"/>
    </node>

    <!-- An external node publishing sensor_msgs/Image on the "image" topic is required -->
</launch>
```

k-okada, 2018-05-17:
I think this similar to https://github.com/ros-perception/image_pipeline/issues/283 and my previous fix https://github.com/ros-perception/image_pipeline/pull/279 was not sufficient. I have confirmed this issue and also https://github.com/yoshito-okada/image_pipeline/tree/fix_image_view_nodelet works perfect

@yoshito-okada could you create new PullRequest  using  https://github.com/yoshito-okada/image_pipeline/tree/fix_image_view_nodelet , so that every one can easily find your solution?

yoshito-okada, 2018-05-17:
I just made a PR #337 .

JWhitleyAStuff, 2019-04-25:
Should be closed by merge of #337.

commit 43081f3d (2019-04-10): Merge pull request #337 from yoshito-okada/fix_image_view_nodelet

Fix threading issue in image_view nodelet. Closes #331.
closed 2019-04-25 by JWhitleyAStuff