ros-perception/image_pipeline #333: Symbol lookup error in image_publisher

stwirth, 2018-05-08:
ROS kinetic
Ubuntu 16.04
```
$ apt-cache policy ros-kinetic-image-publisher ros-kinetic-opencv3 ros-kinetic-cv-bridge
ros-kinetic-image-publisher:
  Installed: 1.12.22-0xenial-20180416-170513-0800
  Candidate: 1.12.22-0xenial-20180416-170513-0800
  Version table:
 *** 1.12.22-0xenial-20180416-170513-0800 500
        500 http://packages.ros.org/ros/ubuntu xenial/main amd64 Packages
        100 /var/lib/dpkg/status
ros-kinetic-opencv3:
  Installed: 3.3.1-5xenial-20180315-113535-0800
  Candidate: 3.3.1-5xenial-20180315-113535-0800
  Version table:
 *** 3.3.1-5xenial-20180315-113535-0800 500
        500 http://packages.ros.org/ros/ubuntu xenial/main amd64 Packages
        100 /var/lib/dpkg/status
ros-kinetic-cv-bridge:
  Installed: 1.12.8-0xenial-20180416-143935-0800
  Candidate: 1.12.8-0xenial-20180416-143935-0800
  Version table:
 *** 1.12.8-0xenial-20180416-143935-0800 500
        500 http://packages.ros.org/ros/ubuntu xenial/main amd64 Packages
        100 /var/lib/dpkg/status
```

When running the `image_publisher` I just get a symbol lookup error:
```
$ rosrun image_publisher image_publisher data/blank_image.png 
/opt/ros/kinetic/lib/image_publisher/image_publisher: symbol lookup error: /opt/ros/kinetic/lib//libimage_publisher.so: undefined symbol: _ZN2cv12VideoCaptureC1Ev
```
It looks like `image_publisher` is missing a link to a library. Maybe OpenCV's `videoio`?

k-okada, 2018-05-17:
@stwirth 
recent update on kinetic deb packages should fix this problem. Please check again.
https://discourse.ros.org/t/new-packages-for-kinetic-2018-05-16/4796

```
k-okada@p51s:~/catkin_ws/ws_imagepipeline$ apt-cache policy ros-kinetic-image-publisher ros-kinetic-opencv3 ros-kinetic-cv-bridgeros-kinetic-image-publisher:
  Installed: 1.12.23-0xenial-20180510-064829-0800
  Candidate: 1.12.23-0xenial-20180510-064829-0800
  Version table:
 *** 1.12.23-0xenial-20180510-064829-0800 500
        500 http://packages.ros.org/ros/ubuntu xenial/main amd64 Packages
        100 /var/lib/dpkg/status
ros-kinetic-opencv3:
  Installed: 3.3.1-5xenial-20180315-113535-0800
  Candidate: 3.3.1-5xenial-20180315-113535-0800
  Version table:
 *** 3.3.1-5xenial-20180315-113535-0800 500
        500 http://packages.ros.org/ros/ubuntu xenial/main amd64 Packages
        100 /var/lib/dpkg/status
ros-kinetic-cv-bridge:
  Installed: 1.12.8-0xenial-20180416-143935-0800
  Candidate: 1.12.8-0xenial-20180416-143935-0800
  Version table:
 *** 1.12.8-0xenial-20180416-143935-0800 500
        500 http://packages.ros.org/ros/ubuntu xenial/main amd64 Packages
        100 /var/lib/dpkg/status
k-okada@p51s:~/catkin_ws/ws_imagepipeline$ rosrun image_publisher image_publisher lena.png [ INFO] [1526539082.850100953]: File name for publishing image is : lena.png
[ INFO] [1526539082.859851236]: Flip horizontal image is : false
[ INFO] [1526539082.860337887]: Flip flip_vertical image is : false
```

stwirth, 2018-05-18:
Confirmed that it works after upgrading. Thanks!

closed 2018-05-18 by stwirth

andresR8, 2018-06-05:
I'm having the same issue.

ROS lunar
Ubuntu 17.10

```
$apt-cache policy ros-lunar-image-publisher ros-lunar-opencv3 ros-lunar-cv-bridge
ros-lunar-image-publisher:
  Installed: 1.12.21-0zesty-20180115-003716-0800
  Candidate: 1.12.21-0zesty-20180115-003716-0800
  Version table:
 *** 1.12.21-0zesty-20180115-003716-0800 100
        100 /var/lib/dpkg/status
ros-lunar-opencv3:
  Installed: 3.3.1-0zesty-20180114-231128-0800
  Candidate: 3.3.1-0zesty-20180114-231128-0800
  Version table:
 *** 3.3.1-0zesty-20180114-231128-0800 100
        100 /var/lib/dpkg/status
ros-lunar-cv-bridge:
  Installed: 1.12.7-0zesty-20180115-001359-0800
  Candidate: 1.12.7-0zesty-20180115-001359-0800
  Version table:
 *** 1.12.7-0zesty-20180115-001359-0800 100
        100 /var/lib/dpkg/status ```