ros-perception/image_pipeline #351: Stereo Calibration from images file

astronaut971, 2018-08-08:
Hi
I have a set of images of a checkerboard pattern from a camera whose calibration info I would like to find out. I have a ROS driver for my Point Grey Chameleon3 Cameras and can publish rostopic list like

```
/camera/left/camera_info
/camera/left/image_raw
/camera/left/image_raw/compressed
/camera/left/image_raw/compressed/parameter_descriptions
/camera/left/image_raw/compressed/parameter_updates
/camera/left/image_raw/compressedDepth
/camera/left/image_raw/compressedDepth/parameter_descriptions
/camera/left/image_raw/compressedDepth/parameter_updates
/camera/left/image_raw/theora
/camera/left/image_raw/theora/parameter_descriptions
/camera/left/image_raw/theora/parameter_updates
/camera/parameter_descriptions
/camera/parameter_updates
/camera/right/camera_info
/camera/right/image_raw
/camera/right/image_raw/compressed
/camera/right/image_raw/compressed/parameter_descriptions
/camera/right/image_raw/compressed/parameter_updates
/camera/right/image_raw/compressedDepth
/camera/right/image_raw/compressedDepth/parameter_descriptions
/camera/right/image_raw/compressedDepth/parameter_updates
/camera/right/image_raw/theora
/camera/right/image_raw/theora/parameter_descriptions
/camera/right/image_raw/theora/parameter_updates
/diagnostics
```
Unfortunately I can not find instructions on how to feed images to the ROS Camera calibration tool instead of the camera itself. Is there any way to do this using the ROS package? So far my understanding is, firstly to rectify the images coming from my camera. So I need to rectify the images that I saved and wonna use later for the Calibration. Is that right? Is that correcet sequence and if the case what after that?Alternatively, are there any other tools I could use? Any help? Thanks

jveitchmichaelis, 2018-10-30:
> Is there any way to do this using the ROS package? 

You should be able to do this with `tarfile_calibration.py`. It's not really documented anywhere, but it should do the job.

https://github.com/ros-perception/image_pipeline/blob/indigo/camera_calibration/scripts/tarfile_calibration.py

Put your images in a tar archive, call them "left_x.png" and "right_x.png". See https://answers.ros.org/question/11537/creating-a-bag-file-out-of-a-image-sequence/ 

Example usage:

`rosrun camera_calibration tarfile_calibration.py MyImages.tar --square 0.108 --size 8x6`

> So far my understanding is, firstly to rectify the images coming from my camera. So I need to rectify the images that I saved and wonna use later for the Calibration. Is that right?

Yes, but the whole point of calibration is to compute the rectification transform. So what goes into the calibrator are raw images. The calibration that you get out is then used to rectify your images.

> Is that correcet sequence and if the case what after that?

Usually you'd then run a `stereo_image_proc` node alongside your camera which subscribes to your raw images and publishes `image_rect` topics. These can then be used by whatever you want, typically stereo matching.

> Alternatively, are there any other tools I could use? 

You can use OpenCV or SimpleCV to perform the calibration yourself, but you'd have to produce the YAML file that ROS requires (which is simple enough).

JWhitleyAStuff, 2019-04-25:
Closing with (assumed) resolution.

closed 2019-04-25 by JWhitleyAStuff

akdhandy, 2022-11-18:
Hello,I have been trying to use this but I always encounter this error.I tried to load a tar file and run it using:

rosrun camera_calibration tarfile_calibration.py archive.tar --square 0.051 --size 7x10.

```Traceback (most recent call last):
  File "/opt/ros/noetic/lib/camera_calibration/tarfile_calibration.py", line 228, in <module>
    cal_from_tarfile(boards, tarname, options.mono, options.upload, calib_flags, options.visualize, options.alpha)
  File "/opt/ros/noetic/lib/camera_calibration/tarfile_calibration.py", line 62, in cal_from_tarfile
    calibrator.do_tarfile_calibration(tarname)
  File "/opt/ros/noetic/lib/python3/dist-packages/camera_calibration/calibrator.py", line 1436, in do_tarfile_calibration
    self.cal(limages, rimages)
  File "/opt/ros/noetic/lib/python3/dist-packages/camera_calibration/calibrator.py", line 1081, in cal
    goodcorners = self.collect_corners(limages, rimages)
  File "/opt/ros/noetic/lib/python3/dist-packages/camera_calibration/calibrator.py", line 1107, in collect_corners
    raise CalibrationException("No corners found in images!")
camera_calibration.calibrator.CalibrationException: No corners found in images!
```