ros-perception/image_pipeline #360: Port Image_pipeline on ROS2

yechun1, 2018-09-30:
If no one started porting image_pipeline on ROS2, I would like to do it in following weeks.

yechun1, 2018-10-16:
Could image_pipeline's owner/maintainer please help to create ros2 branch for [ros-perception/image_pipeline](https://github.com/ros-perception/image_pipeline)?  

The depth_image_proc has been ported on ROS2, need help to review, here is the commit: d5eef39. Thanks.

yechun1, 2018-10-23:
@mjcarroll thanks for help to create ros2 branch, first PR to enable ros2 depth_image_proc is ready on #362.

jacobperron, 2019-09-10:
Any update on `stereo_image_proc`?

SteveMacenski, 2019-09-10:
I can tell you @JWhitleyAStuff and I aren't thinking about it. I had alot of the ROS2 work in this report scheduled post-ROSCon as I'm a little preoccupied with Navigation2 and slam toolbox. 

@jacobperron is that something you're motivated to look at?

yechun1, 2019-09-10:
I tried to port stereo_image_proc in the beginning of the year but blocked by image_proc which is not ready then. And now the image_proc has been ported and stereo_image_proc porting could be continue, but very sorry that I have no resource on this right now. 
It would be glad if some one could have a time to work on it. Below is previously patch(not fully completed) just for reference.

port ros2 stereo_image_proc: https://github.com/yechun1/image_pipeline/commit/e9993bed6220d28488fa73516eb95110d68f804e
on (https://github.com/yechun1/image_pipeline/tree/ros2_stereo_image_proc)

klintan, 2019-09-11:
@yechun1 one version of image_proc is actually already merged, unfortunately it is not possible to run it as a component yet, but I'll put in some PRs for that. However I think it might still be considered "ready/done" for image_pipeline. (at least as a initial release candidate)

jacobperron, 2019-09-11:
> @jacobperron is that something you're motivated to look at?

@SteveMacenski I don't have the bandwidth at the moment, I'll let you know if things change.

SteveMacenski, 2019-09-12:
@jacobperron that doesn't surprise me in the slightest :wink:

jacobperron, 2019-12-10:
FYI, I have some bandwidth now and will start porting `stereo_image_proc`.

laszloturanyi, 2019-12-11:
Hi,
I managed to port stereo_image_proc some days ago. There is still some work to do to pass all the tests but now it can function.

@jacobperron you do not have to do it from scratch I rebased it to your jacob/ros2_omage_processor
you can find a link to my [fork] (https://github.com/laszloturanyi/image_pipeline/tree/ros2_stereo_image_proc)

I would be glad if you could have a look at it I have limited capacity to finish this but maybe we can find a way to do this efficiently.

jacobperron, 2019-12-11:
@laszloturanyi Thanks for sharing your effort! I'm happy to build on your fork if you don't have the time to finish the port.

laszloturanyi, 2019-12-12:
@jacobperron that would be great. I added you as collaborator.

jacobperron, 2019-12-13:
See #486 for a port of stereo_image_proc.

@laszloturanyi I ended up not basing my fork on yours, as I had already done some of the same preliminary refactoring as you. It would be great if you could help review/try out the proposed port :)

I'll be adding support for parameter limits/reconfiguring shortly.

closed 2020-04-10 by SteveMacenski