ros-perception/image_pipeline #371: ros2 travis ci build fail

yechun1, 2018-12-07:
PR for ROS2 will make CI build fail, It seems like the .travis.yaml does not support ROS2 crystal.

for one build log:
https://travis-ci.org/ros-perception/image_pipeline/builds/463164488?utm_source=github_status&utm_medium=notification

```
$ git clone --depth=50 https://github.com/ros-perception/image_pipeline.git ros-perception/image_pipeline
Cloning into 'ros-perception/image_pipeline'...
remote: Enumerating objects: 719, done.
remote: Counting objects: 100% (719/719), done.
remote: Compressing objects: 100% (374/374), done.
remote: Total 719 (delta 418), reused 563 (delta 310), pack-reused 0
Receiving objects: 100% (719/719), 281.53 KiB | 6.55 MiB/s, done.
Resolving deltas: 100% (418/418), done.
$ cd ros-perception/image_pipeline
0.45s$ git fetch origin +refs/pull/369/merge:
remote: Enumerating objects: 193, done.
remote: Counting objects: 100% (141/141), done.
remote: Compressing objects: 100% (29/29), done.
remote: Total 68 (delta 38), reused 63 (delta 36), pack-reused 0
Unpacking objects: 100% (68/68), done.
From https://github.com/ros-perception/image_pipeline
 * branch            refs/pull/369/merge -> FETCH_HEAD
$ git checkout -qf FETCH_HEAD
Setting environment variables from .travis.yml
$ export OPENCV_VERSION=2
$ export ROS_DISTRO=indigo
```

klintan, 2019-02-14:
It's still running tests for all branches using ROS Indigo right?

JWhitleyAStuff, 2019-04-08:
See #396 for discussion.

SteveMacenski, 2019-04-08:
@JWhitleyAStuff isn't this referencing ROS2 though?

JWhitleyAStuff, 2019-04-08:
Closed too quickly. Mis-read.

JWhitleyAStuff, 2019-04-25:
@yechun1 and @klintan - CI on the primary branches (`indigo`, `kinetic`, and `melodic`) has been transitioned to CircleCI. I asked about this in #369 but, if you two are OK with it, I would like to rebase the `ros2` branch on `melodic`. This will pull in the ability to use CircleCI and I can modify the CI files to use a ROS2 docker image.

klintan, 2019-04-25:
> @yechun1 and @klintan - CI on the primary branches (`indigo`, `kinetic`, and `melodic`) has been transitioned to CircleCI. I asked about this in #369 but, if you two are OK with it, I would like to [...]

I'm good with that :) thanks!

JWhitleyAStuff, 2019-04-25:
Implemented (and fixed) with #408.

PR #408 (2019-04-25): Maint/ros2 ci
Closes #371. Adds basic CI for ROS Crystal. Official Docker images are not yet available for ROS Dashing - will be added once they are.
closed 2019-04-25 by JWhitleyAStuff