ros-perception/image_pipeline #377: RuntimeError: cv_bridge.cvtColorForDisplay() while trying to convert image from '8UC3' to 'bgr8' an exception was thrown ([8UC3] is not a color format. but [bgr8] is.

zhudaoruyi, 2019-01-09:
```sh
nvidia@ubuntu:~/catkin_csi_camera$ rosrun image_view extract_images_sync _inputs:='[/csi_cam/image_raw]'
[ERROR] [1547019957.683194]: bad callback: <bound method Subscriber.callback of <message_filters.Subscriber object at 0x7f54800390>>
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/message_filters/__init__.py", line 75, in callback
    self.signalMessage(msg)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/message_filters/__init__.py", line 57, in signalMessage
    cb(*(msg + args))
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/message_filters/__init__.py", line 224, in add
    self.signalMessage(*msgs)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/message_filters/__init__.py", line 57, in signalMessage
    cb(*(msg + args))
  File "/home/nvidia/catkin_csi_camera/src/image_view/scripts/extract_images_sync", line 89, in save
    do_dynamic_scaling=self.do_dynamic_scaling)
RuntimeError: cv_bridge.cvtColorForDisplay() while trying to convert image from '8UC3' to 'bgr8' an exception was thrown ([8UC3] is not a color format. but [bgr8] is. The conversion does not make sense)

```

duchengyao, 2019-03-18:
just remove some code from `image_view/scripts/extract_images_sync`

https://github.com/ros-perception/image_pipeline/pull/378/commits/fe4f530129d17b3d8d95641fc38e466c6b1b369e

JWhitleyAStuff, 2019-04-25:
It looks like a fix for this issue has already been applied. Closing.

closed 2019-04-25 by JWhitleyAStuff

xkaraman, 2022-04-01:
> It looks like a fix for this issue has already been applied. Closing.

The fix as far as i am aware never merged into neither melodic, or neotic.

HaiCLi, 2024-09-03:
> > It looks like a fix for this issue has already been applied. Closing.
> 
> @JWhitleyWork The fix as far as i am aware never merged into neither melodic, or noetic.
> 
> After applying the patch explained above, both rgb and depth are saved but encodings are probably mixed up and images are not as the originals.

Hi,
Did you solve it? I found in Ros2 the same problem when I try to extract image from video in ros2bag.