ros-perception/image_pipeline #390: image_view output dailog box issue

vishalgpt579, 2019-03-06:


JWhitleyAStuff, 2019-04-08:
Which version of ROS are you using and on which version of Ubuntu?

vishalgpt579, 2019-04-18:
> Which version of ROS are you using and on which version of Ubuntu?

@JWhitleyAStuff  I'm using following version of ROS  :

 >rosdistro: melodic
 >rosversion: 1.14.3

 And following version of Ubuntu:

> Distributor ID:	Ubuntu
>Description:	Ubuntu 18.04.2 LTS
>Release:	18.04
>Codename:	bionic

JWhitleyAStuff, 2019-04-18:
@vishalgpt579 - What is the original pixel height/width of the image you're trying to display? It looks like the default for `image_view` is to provide a window with 1:1 pixel size of the original image. If the image is larger than your desktop size, what you're seeing would be the result. There are a few ways around this, but they all require changes to the node.

vishalgpt579, 2019-04-22:
@JWhitleyAStuff, The original pixel height/width of the image is 1280x1024(which is same as of my monitor). As i mentioned in original question , that lowering the resolution provides a smaller windows , i understand it now, which is because of this reason...
>It looks like the default for image_view is to provide a window with 1:1 pixel size of the original image

Could you please describe those ways of sorting this out @JWhitleyAStuff ?

JWhitleyAStuff, 2019-04-22:
Here are a couple of options that would work:

1. Provide a "fullscreen" parameter which is a Boolean. If this parameter is "true," then set the output window to be full-screen.
2. Provide a parameter (difficult to think of a name) that, if set to "true," means that if the output image is larger than the desktop area, set the output window to full-screen.
3. Provide a "output scaling" parameter which defaults to 1.0. The output image window would then be the original width and height multiplied by this value.

I think number 3 would solve your issue in the most convenient way but I'm open to suggestions and feedback.

SteveMacenski, 2019-04-22:
I'd even recommend a 4th one, where the size of the image is taken by the size of the window. Allow the window to rescale anyway it wants then just have the image scale to the smallest dimension

JWhitleyAStuff, 2019-04-23:
@vishalgpt579 - PR #389 gives the full-screen option (my option 1) but I'm guessing this doesn't really solve your use-case, correct?

vishalgpt579, 2019-04-24:
@JWhitleyAStuff @SteveMacenski , thank you for your recommendations. But since I'm just a beginner, all i can infer from your suggestion is either one of the 2 things ...

1) To edit the [launch file ](https://github.com/lrse/whycon/blob/master/launch/whycon.launch) which I'm running , instead of the line 
`rosrun image_view image_view image:=/visionSensor/image_rect _autosize:=True`
and somehow make changes to   _output ="screen"_  on line 14 of [whycon.launch](https://github.com/lrse/whycon/blob/master/launch/whycon.launch)

2) Secondly , To edit the image_view package itself. 

Either of them i have not much of a clue...  could you guild me further?
Please correct me if I'm wrong .

vishalgpt579, 2019-04-25:
> @vishalgpt579 - PR #389 gives the full-screen option (my option 1) but I'm guessing this doesn't really solve your use-case, correct?

@JWhitleyAStuff ,I tried doing this... 
`rosrun image_view image_view image:=/visionSensor/image_rect _autosize:=True _fullscreen:=False`

Yes, you were correct. It still doesn't works. 

[image]

Output  remained the same ... unresizable 

[image]

closed 2019-05-02 by vishalgpt579

lamhung81, 2025-06-06:
Did you successfully resolve this problem?