ros-perception/image_pipeline #393: image_pipeline Repository Maintenance

JWhitleyAStuff, 2019-04-01:
There are currently many open and unanswered PRs and Issues in this repo. I have tried multiple times to contact the maintainers with no response (see https://discourse.ros.org/t/perception-sig-maintainers-still-maintaining/7065/15). @SteveMacenski and myself would be interested/able to take over maintenance of this repo if the current maintainers are not interested/able to maintain it any longer.

Thanks,
Josh Whitley

JWhitleyAStuff, 2019-04-02:
@knorth55 - The convention is to have at least one responsible maintainer, especially for a package that is actively developed. The Orphaned Package Maintainer's group is for packages which the community believes will not see active maintenance, just continued releases for current and future versions of ROS. From the wiki page you linked:

    A package lacking active maintenance, specifically one which is released in previous ROS distro but not released in the latest ROS distro.

and

    A team of volunteers that will do ONLY the release job, to keep releases of existing (and useful) ROS packages in new ROS distros with a minimum of community effort.

knorth55, 2019-04-03:
@JWhitleyAStuff of course, there will be one responsible maintainer from the ros orphaned package.
Releasing means that fixing bug and releasing the package properly.
I think this is what you want to do.
In other word, you should not add new features in this package even if you will be a maintainer of this package because many projects depend on this package.

JWhitleyAStuff, 2019-04-04:
Thank you to @jkammerl for granting me admin access to this repo! I'll do my best to get it cleaned up soon!

JWhitleyAStuff, 2019-04-04:
@knorth55 Active contributions to this repository are very much encouraged, we just have to be very careful about thorough testing.

closed 2019-04-04 by JWhitleyAStuff

jkammerl, 2019-04-04:
Thanks for escalating this and even more importantly for offering your support! I agree that this is unfortunate and that your valuable contributions shouldn't be affected by the lack of maintainer support. Given that nobody from the official maintainers responded now for a couple of days, I went ahead and gave admin rights to @SteveMacenski and @JWhitleyAStuff. Your maintainer support is greatly appreciated but with great powers comes... you know it :)

JWhitleyAStuff, 2019-04-04:
@jkammerl - Do you have the ability to give us access to image_pipeline-release for releases as well?

knorth55, 2019-04-05:
i strongly disagree to encourage new feature addition tonthis repository bacause many robots uses `image_pipeline`.
I really want you to just fix bugs and release this package.

SteveMacenski, 2019-04-05:
@knorth55 adding new features doesn't mean gutting existing ones. I'm not sure why making a new `awesome-image-processing-thingy` would effect `crop` nodelet, for instance. New things should not break existing things. Existing things just need bug fixes

SteveMacenski, 2019-04-05:
@jkammerl I was wondering if I could be added to the larger ros-perception organization? I would have interest in supporting and maintaining more than just this (in fact, I'm probably better suited to alot of the other stuff in this org). In particular:
- open_karto / slam_karto
- open_gmapping / gmapping
- SBA
- perception_pcl
- imu pipeline 

I hate to see so many of these things I know and love sit idle.

knorth55, 2019-04-05:
@SteveMacenski 
new features are good but you can add it in new repo.
i want to make this package stable because it might be side effects to many packages.
this package is fundamental package and need to work as it is now.

JWhitleyAStuff, 2019-04-08:
@knorth55 - Just because something is relied upon doesn't mean it shouldn't evolve. We will continue to do as the community decides, not just a single voice. We will do our best to maintain backward-compatibility and notify the community if there are any breaking changes coming, but the community does not stop evolving just because something they built has come to be relied upon. I'm sorry if your infrastructure relies on internal details of this package functioning as it does now but, if this is the case, it is simply bad design. You should never assume that any code outside your own will continue to function the same internally or have exactly the same number of external interfaces (though you can usually assume that the existing external interfaces or APIs will remain stable until a new major release). This is just how software development works. Without this premise, nothing would ever change.

SteveMacenski, 2019-04-08:
I think @knorth55 might have some fundamental misconceptions about how things may change. No one's talking about completely changing the way an existing nodelet is done, that would be crazy. However, adding new nodelets to do new things, and patching bugs in existing ones, or extending them to work with NVIDIA/Windows platforms is.

There's again no reason why `new-awesome-thing` would effect `crop` nodelet if you rely on it. 

This repo and organization isn't a museum, it lives and breathes as new capabilities and needs arise. Especially with mature and robust additions to projects like openCV.