ros-perception/image_pipeline #396: Fix travis CI

SteveMacenski, 2019-04-05:


JWhitleyAStuff, 2019-04-08:
@SteveMacenski - For the repositories to which I have maintenance responsibilities, I have been switching to CircleCI which is more flexible, updated considerably more often, and has much lower queue times (see https://github.com/ros-drivers/velodyne/blob/master/.circleci/config.yml as an example). Any opposition?

SteveMacenski, 2019-04-08:
None at all. I haven't used it personally, but that's not a reason _not_ to learn something :)

JWhitleyAStuff, 2019-04-08:
Thanks. I'll get on it as soon as I'm back from lunch. :)

JWhitleyAStuff, 2019-04-08:
Fixed in #398 by changing to CircleCI and using new build setup. Since Indigo recently EOL'd, we're only running CI on Kinetic and Melodic branches.

closed 2019-04-08 by JWhitleyAStuff