ros-perception/image_pipeline #397: No point cloud published while using SGM for matching

MattDom, 2019-04-06:
If I use the BM algorithm to match my synchronized images from two Point Grey cameras, I can get a point cloud as expected. However, when I use the SGM algorithm configured with dynamic_reconfigure, the Point Cloud theme stops working immediately.

If I change the algorithm back to BM, the point cloud will be published again. Checking the source code didn't brought me any new ideas, where the problems comes from.

I use ROS Kinectic with Ubuntu 16.04.

AJwgnr, 2019-04-08:
Are you getting any errors in your console?

Could be that the SGM algorithm computation is not fast enough on your system. You could e.g. decrease the resolution of your images to check whether it depends on the computational power.

SteveMacenski, 2019-04-08:
Please make sure to read through http://wiki.ros.org/stereo_image_proc/Tutorials/ChoosingGoodStereoParameters

It should help, I think what you might be seeing is something described in the tuning guide of producing no points

MattDom, 2019-04-08:
@SteveMacenski Thanks for your response. I read through the tutorial previously. The fixed frame and everything is set correctly and I see the point cloud when I use the Blockmatch-algorithm. Further, the disparity image is calculated properly with the SGM-algorithm, just the point cloud is not published. For my understanding, the calculation is of the point cloud should not depend on the matching-algorithm. 

@AJwgnr Thanks for your message, too. As stated above, the SGM produces disparity maps, with ca. 1 fps. So, the post-processing for the point cloud should be the same as for the Blockmatching-algorithm. Further, I got no error messages, which would be a bit strange, if it is really an error.

Thanks for your support.

SteveMacenski, 2019-04-08:
The pointcloud is directly related to the stereo matching, without matched pairs, a disparity map cannot be generated, which is in turn used to generate a pointcloud. When you're tuning this, is there still a disparity map?

MattDom, 2019-04-08:
> The pointcloud is directly related to the stereo matching, ...  

This was a missunderstanding. Of course, it is important for getting the disparity map, but the calculation for the point cloud depends on the resulting disparity map and not on the algorithm. Hope, I could put it in a nutshell 😄 

> When you're tuning this, is there still a disparity map?

Yes, thats the point, where I got stuck...

SteveMacenski, 2019-04-08:
Can you post an example set of disparity maps as images? Maybe there's no real pair sets. Combine that with looking at `rostopic hz /pts/topic` to make sure its not just publishing a super sparse message

MattDom, 2019-04-08:
Yes, sure. I am already out of the lab for today (8 hours ahead), but tomorrow this is my first job.

SteveMacenski, 2019-04-09:
The frequency of what is 1fps to 5fps? Input, disparity, point cloud? 

Is the pointcloud being published at all? Can you check to see how many points are in it? 
I think with that we should have enough to make a diagnosis

MattDom, 2019-04-10:
Sorry, the frequency of the disparity maps is ca. 1fps, when I use the SGM and about 5fps for the BM-algorithm. 

The point cloud is only published, when I use the Blockmatching and not for the SGM. The point cloud, I get from the Blockmatching, contain around 9k to 1M points. So, with the right settings it is pretty dense.

@SteveMacenski: Thank you very much for your patience and your help.

SteveMacenski, 2019-04-10:
So you're saying, to clarify, that it is being published in the SGM case, but its empty? Or just sparse? Given these are different methods, did you try reverting to  defaults to see if it publishes with SGM?

MattDom, 2019-04-12:
Okay. Probably I was a bit unprecisely. I'm sorry for that.

1. Case --> Blockmatching

    -  Disparitymaps come with ca. 5fps
    - Pointcloud is published with about 3fps
    - Everything works as expected

2. Case --> SGM

    - Disparitymaps come with ca. 1fps
    - No point cloud is published
    - There is no pc-message, no even an empty
    - Post-processing the disparitymap to a point cloud in Matlab works like charm

Is it possible, that the point cloud won't be calculated with the SGM due to the low fps-rate? My guess comes from the point, that the function to calc the point cloud also needs the camera info. So, maybe it is just a sync-problem and can be solved be a larger queue...?

SteveMacenski, 2019-04-15:
Did you try reverting to default configurations for SGM and does it publish a pointcloud then? If not, might be something off. If it does publish, then its potentially the parameters chosen

MattDom, 2019-04-16:
Okay, it works. I changed nothing, honstely. I booted the whole thing this morning and it worked. Same parameters, same hardware, everything unchanged. Unfortunatly, I can't determine where or what was the error. 

However, I will try to find what went wrong and post it then.

@SteveMacenski: Thank you for the support.

closed 2019-04-16 by MattDom

SteveMacenski, 2019-04-16:
Bam. All we need to do is ask dumb questions and complex issues crumble at our feat!