ros-perception/image_pipeline #402: disparity image is displayed but pointcloud shows only a small dot in rviz

Syed-Izzat-Ullah, 2019-04-16:
Hi all,
I am trying to get `disparity image` and `pointclouds` from `VREP's` two `vision sensors`(assembled like stereo camera) using `Stereo_Image_Proc.` Disparity image is successfully displayed and pointcloud topic (i-e `/points2`) publishes consistent data at each iteration(I mean its data contains few numbers like 197, 172,0 ,0,0,0 etc repeatedly). I don't know where am I doing mistake. I would be highly grateful if someone please solve my this problem. I have attached `rqt_graph`, `rostopic echo /points2` and `rviz` display o
[image]
[image]
[image]

f pointcloud. Thanks

SteveMacenski, 2019-07-10:
This is a configuration or visualization issue. See links below to help, closing.

https://github.com/ros-perception/image_pipeline/issues/397
http://wiki.ros.org/stereo_image_proc/Tutorials/ChoosingGoodStereoParameters

closed 2019-07-10 by SteveMacenski

Mdmorshadurrahman, 2022-04-19:
> Hi all, I am trying to get `disparity image` and `pointclouds` from `VREP's` two `vision sensors`(assembled like stereo camera) using `Stereo_Image_Proc.` Disparity image is successfully displayed and pointcloud topic (i-e `/points2`) publishes consistent data at each iteration(I mean its data contains few numbers like 197, 172,0 ,0,0,0 etc repeatedly). I don't know where am I doing mistake. I would be highly grateful if someone please solve my this problem. I have attached `rqt_graph`, `rostopic echo /points2` and `rviz` display of pointcloud. Thanks [image] [image] [image]
> 
> EDITED: when i convert the pointcloud2 to image using below two commands I can see the image from pointcloud data. the problem is i can't visualize it in rviz. `rosrun pcl_ros convert_pointcloud_to_image input:=/points2 output:=/my_image` `rosrun image_view image_view image:=/my_image`

Hi

> Hi all, I am trying to get `disparity image` and `pointclouds` from `VREP's` two `vision sensors`(assembled like stereo camera) using `Stereo_Image_Proc.` Disparity image is successfully displayed and pointcloud topic (i-e `/points2`) publishes consistent data at each iteration(I mean its data contains few numbers like 197, 172,0 ,0,0,0 etc repeatedly). I don't know where am I doing mistake. I would be highly grateful if someone please solve my this problem. I have attached `rqt_graph`, `rostopic echo /points2` and `rviz` display of pointcloud. Thanks [image] [image] [image]
> 
> EDITED: when i convert the pointcloud2 to image using below two commands I can see the image from pointcloud data. the problem is i can't visualize it in rviz. `rosrun pcl_ros convert_pointcloud_to_image input:=/points2 output:=/my_image` `rosrun image_view image_view image:=/my_image`

Need you help ! Got stuck in the same situation badly.

JMadhup, 2022-06-28:
@syediu Were you able to solve this issue?

syediu, 2022-06-28:
@Mdmorshadurrahman @JMadhup Please apology my late response. Yes, I did solve the issue.
**Solution:**
While using the `stereo camera`, you also need to supply the `baseline` between the left and right camera, if you don't specify this, you will not be able to get depth. See [this link](http://docs.ros.org/en/noetic/api/sensor_msgs/html/msg/CameraInfo.html#:~:text=For%C2%A0a%C2%A0stereo,is%C2%A0given%C2%A0by%3A) for reference, in the [Projection matrix](http://docs.ros.org/en/noetic/api/sensor_msgs/html/msg/CameraInfo.html#:~:text=Projection/camera%C2%A0matrix), supply the value of the correct `baseline` as mentioned in the link.

[Projection matrix](http://docs.ros.org/en/noetic/api/sensor_msgs/html/msg/CameraInfo.html#:~:text=Projection/camera%C2%A0matrix) can be found in the [Sensor_msgs/CameraInfo](http://docs.ros.org/en/noetic/api/sensor_msgs/html/msg/CameraInfo.html) when publishing the camera information.

JMadhup, 2022-06-30:
@syediu are you referring tto the <hackBaseLine> attribute of the plugin?