ros-perception/image_pipeline #403: Update image_pipeline to use new ros2 composition model

mjcarroll, 2019-04-19:
In `dashing`, we have updated how ROS2's components work.  This allows developers to write a `rclcpp::Node` that can be used either in standalone or composed into a larger executable, and defer that decision until runtime.  It is essentially a replacement for ROS1's `nodelet` concept, but without having to do special implementations.

I believe that `image_pipeline` would be a good candidate to take advantage of this new model, as it closely mirrors the way that `nodelets` were used in ROS1.

References:
* https://github.com/ros2/rclcpp/tree/master/rclcpp_components
* https://github.com/ros2/demos/tree/master/composition
* Documentation (WIP) https://github.com/ros2/ros2_documentation/pull/168
* Composable node launch actions: https://github.com/ros2/launch_ros/pull/9

JWhitleyAStuff, 2019-04-25:
While Dashing has "technically" been released and I really like this feature (been looking forward to it for a long time), I don't think it's well-supported enough yet (e.g. no official Docker images, `cv_bridge` not yet released, `image_geometry` not yet released, etc.) to target it in this repo. Long-term, though, this makes a lot of sense.

SteveMacenski, 2019-04-27:
I disagree, I heard William talk about this at an event last week and its particularly useful for this repo and other sensor pipeline metapackages. Nodelets don't exist in ROS2 and this gives you the zero copy in-process capabilities. 
None of the ROS2 releases are LTS and should be considered deprecated when a new one comes out every 6 months and there's demos working today.

SteveMacenski, 2019-11-03:
@mjcarroll are you imagining image_proc or all the image pipeline nodes?

JWhitleyAStuff, 2019-11-03:
I've been doing a lot of porting recently and my earlier statement no longer applies. I think it makes sense to do all of them.

SteveMacenski, 2019-11-03:
OK.

What are your porting plans / actions?

JWhitleyAStuff, 2019-11-03:
Luckily, the port isn't that complicated. We just need to make sure that:

1. Each node is a separate class which inherits from `rclcpp::Node` and takes an `rclcpp::NodeOptions` object.
2. An executable is generated for each `component` (this is just an option in the call to `rclcpp_components_register_node`).
3. Each node is registered as a `component` using the appropriate CMake macros.
4. Some example pipelines are provided either as launch files or binaries which combine several components.

SteveMacenski, 2019-11-03:
Got it, I thought you meant you were in the middle of some of these ports.

There's definitely still alot to be cleaned up (and some to be done to get image view & stereo pipeline back up). Mostly around the configurations for updating & consistency.

SteveMacenski, 2019-11-04:
I just did a quick look over, it appears they're all registered as components expect the missing ports

JWhitleyAStuff, 2019-11-04:
Cool. Can probably close this then, since it's an established pattern.

closed 2019-11-04 by SteveMacenski