ros-perception/image_pipeline #404: List tickets/PRs to merge and freeze for release

SteveMacenski, 2019-04-19:
We've merged a bunch of stuff recently. We should come up with a list of those we want to get resolved before a new release and do some system level tests to ensure nothing is broken prior to a new release.

JWhitleyAStuff, 2019-04-23:
For `indigo` and `kinetic`, the following PRs were merged prior to requiring new changes to be against `melodic` unless a back-port was required and would be included in a 1.13 release (I want to do one last release for `indigo` since we already merged these):

- #343 
- #392 
- #395 
- #379 
- #394 
- #337 

In addition, the following PRs were merged only to `kinetic` and would be included in a 1.13 release:

- #401

JWhitleyAStuff, 2019-04-25:
As far as PRs that I would like to merge, here are some (haven't finished reviewing all of them yet):

- #391 
- #389 
- #358 
- #356 (backport to at least `kinetic`)
- #355 
- #352 (backport to `kinetic`)
- #341

JWhitleyAStuff, 2019-04-25:
@SteveMacenski / @mjcarroll - Finally got through the full list of PRs. Thoughts on the list above?

Working my way through issues backlog now.

SteveMacenski, 2019-04-27:
Given Indigo is now EOL, I think we should just leave that as is. I don't want to break peoples' stuff on an EOL distro as I would greatly like to never think about indigo again. I'd just focus on Kinetic / Melodic for the time being given the sheer number of things we've done in the last month.

JWhitleyAStuff, 2019-04-27:
Fair. Do you think we should roll back the PRs on the Indigo branch to the last release so we don't have a disparity between the last release for Indigo and the source version?

SteveMacenski, 2019-04-27:
That's a good point you bring up, I had not thought about that. From a cleanliness perspective I think that would be good since its a pretty easy process.

JWhitleyAStuff, 2019-04-29:
I've hard-rolled-back `indigo` to the version that was released as 1.12.23.

Do you have any thoughts on the list I made of those I would like to merge to `melodic`?

SteveMacenski, 2019-05-01:
I'm OK with the suggested tickets. The ones still open are mostly still failing CI or not super responsive folks. Do we want to wait to get them in?

JWhitleyAStuff, 2019-05-01:
I think I'm OK with calling a hold for now to do testing and release for `kinetic` and `melodic`. Shall we set a target date to complete testing?

JWhitleyAStuff, 2019-05-22:
@SteveMacenski - I've completed pretty thorough testing on `image_view`. Should we split up testing among the nodes that have changed since the last release? The following are files that got changed and need testing:

[] camera_calibration/nodes/cameracalibrator.py
[] camera_calibration/scripts/tarfile_calibration.py
[] camera_calibration/src/camera_calibration/calibrator.py
[] camera_calibration/src/camera_calibration/camera_calibrator.py
[] image_publisher/src/nodelet/image_publisher_nodelet.cpp
[] image_view/source/nodelets/image_nodelet.cpp
[] image_view/src/nodes/extract_images.cpp
[] image_view/src/nodes/image_view.cpp
[] image_view/src/nodes/video_recorder.cpp
[] stereo_image_proc/include/stereo_image_proc/processor.h
[] stereo_image_proc/launch/stereo_image_proc.launch

JWhitleyAStuff, 2019-06-12:
@SteveMacenski - Given my current workload, I know I won't get much testing time here. I'm going to go ahead and release for Melodic and do the testing on Kinetic as time allows. I believe that if we get feedback on Melodic, we can iterate more quickly without much backlash. As time allows, I'll attempt to test the above list on Kinetic further.

SteveMacenski, 2019-07-10:
Hi,

Sorry about disappearing. I was on vacation for 5 weeks and just got back (and now in a new job). Well it seems you released about a month ago and no one came screaming so I take that as a successful release. We should still test. Let me circle back to this in early August. My new job is ~85% open source so I should have cycles to take a more active hand but there's a bunch of logistics that need to be worked out first. 

Going forward, lets not merge PRs that haven't been independently tested by someone that didn't write it. This is a foundational meta-package that touches alot of stuff and I don't want people afraid of running edge.

JWhitleyAStuff, 2019-07-11:
@SteveMacenski - Don't worry about being gone. Totally understandable. And I'm with you on not merging independently-untested PRs but we were kind of in a bind. `melodic` was unusable prior to this release (`image_view` didn't even work) and people were asking for it and neither of us had time to do the independent testing. In the future, I'll do my best to stick to this policy but I think it was a necessary evil this time around.

SteveMacenski, 2019-07-11:
Agreed, I think we needed to get  stuff out but now that its settled down largely, lets take more care

closed 2019-09-03 by SteveMacenski