ros-perception/image_pipeline #407: Replace boost with std as much as possible

lucasw, 2019-04-25:
180 occurrences of `boost::` in image_pipeline.

shared_ptrs, mutexes, lock_guards, threads, const_pointer_casts look like easy changes.

Not sure about boost filesystem or format.

boost::bind can't be replaced without ros callbacks supporting it?  (I'd like to hear that is already on a roadmap- if not maybe taking the initiative here and in other packages will help raise the issue for the core ros packages)

Maybe target this for a post-melodic release.

SteveMacenski, 2019-11-03:
Looking over the ROS2 side of things, the only places boost exists are in areas without ports. I'm personally of the opinion an effort like that in ROS1 would be probably not a good use of time, so I'm leaning towards interpreting this ticket as relating to ROS2 as we move forward unless there's a strong reason to include this in Noetic.

JWhitleyAStuff, 2019-11-03:
Since Boost is required by the framework in ROS1, it's no extra bloat to use it there. However, since it isn't necessary in ROS2, this makes sense.

mikeferguson, 2024-01-19:
Rolling currently has 13 boost instances - one foreach loop in a test, and 4 viewers that use boost/format
```
./image_proc/test/rostest.cpp:#include <boost/foreach.hpp>
./image_view/src/image_view_node.cpp:#include <boost/format.hpp>
./image_view/src/stereo_view_node.cpp:#include <boost/format.hpp>
./image_view/src/extract_images_node.cpp:#include <boost/format.hpp>
./image_view/src/image_saver_node.cpp:#include <boost/format.hpp>
./image_view/include/image_view/image_saver_node.hpp:#include <boost/format.hpp>
./image_view/include/image_view/image_saver_node.hpp:  boost::format g_format;
./image_view/include/image_view/image_view_node.hpp:#include <boost/format.hpp>
./image_view/include/image_view/image_view_node.hpp:  boost::format filename_format_;
./image_view/include/image_view/extract_images_node.hpp:#include <boost/format.hpp>
./image_view/include/image_view/extract_images_node.hpp:  boost::format filename_format_;
./image_view/include/image_view/stereo_view_node.hpp:#include <boost/format.hpp>
./image_view/include/image_view/stereo_view_node.hpp:  boost::format filename_format_;
./image_view/package.xml:  <depend>libboost-dev</depend>
```