ros-perception/image_pipeline #416: stereo_image_proc Output-Timing is async

TillBeemelmanns, 2019-05-09:
I am using stereo image proc to convert the raw output of a stereo camera. Both cameras use a Bayer 8 Bit encoding. As shown in the image, the raw messages arrive approximately at the same time, but the converted color images have a time difference of approx. 10 ms.

[image]

That is how I run stereo_image_proc:
`<node name="stereo_image_proc" pkg="stereo_image_proc" type="stereo_image_proc" ns="sensors/camera/" output="screen"></node> `

Has anyone an idea to solve this problem ?

Best Regards

SteveMacenski, 2019-05-09:
Are you asking why it takes 10ms to convert the images? Can you clarify what you mean by "time difference"? Is this is headers / receipt time of another client of the converted images, etc. There's some CPU time associated with that conversion I imagine

TillBeemelmanns, 2019-05-09:
With "time difference" I mean the difference between `right/image_color` msgs and `left/image_color` msgs. They both should be send at the same time, since the raw msgs arrive at the same time.

SteveMacenski, 2019-07-10:
Clarifying question: they arrive 10ms apart, but the timestamps are correct, yes? You should always be using the timestamp in the header of the messages for processing data in ros not current time the message is received. There is no promise that data will be received "fast"* in ROS, the larger the message the more time it may take to come through the pipe into another process. 

On the hand-wavy side of things, 10ms seems large, but you also didnt tell me the resolution of that image. If its something an HD or higher image, there may be some delay. Also keep in mind it take some time to do the calculations you're asking the nodelet to accomplish. 

* ei bounded spec for which all messages are promised to be delivered by

mikeferguson, 2024-01-19:
Closing due to lack of response

closed 2024-01-19 by mikeferguson