ros-perception/image_pipeline #420: Package build fails

acxz, 2019-06-12:
I am trying to build to latest release version of the package 2.0.0. Problems I am encountering right now include having a `signals` dependency. I see that this has been fixed on master, but the fix has not been released. It would be great if this could be released.
After manually making the change, another issue I am encountering is the following:
```
/usr/bin/ld: warning: libopencv_core.so.4.1, needed by /usr/lib/libopencv_videoio.so.4.1.0, may conflict with libopencv_core.so.3.4
/usr/bin/ld: CMakeFiles/video_recorder.dir/src/nodes/video_recorder.cpp.o: in function `callback(boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&)':
video_recorder.cpp:(.text+0x29f): undefined reference to `cv::VideoWriter::open(cv::String const&, int, double, cv::Size_<int>, bool)'
collect2: error: ld returned 1 exit status
```
Looking around I have noticed that a PR has been merged for opencv 4 support but again it seems to not be in the latest release.
Can these fixes be released, so that the package can build successfully?

JWhitleyAStuff, 2019-06-12:
@acxz - The 2.0.0 release is specifically for ROS2. Are you building this in a ROS2 environment or ROS1? I'm just now releasing 1.13.0 for ROS1 Melodic and it should be available on the repos in a couple of weeks at the most.

acxz, 2019-06-12:
I see, my bad. Yes I am using this package for melodic. I noticed that 1.13.0 has been released and I have tried building that. With the latest release (1.13.0), the signal issue has gone away. But I am still getting the same issue with the opencv warning.

acxz, 2019-06-12:
Some more helpful information:
I am trying to build image_view specifically.
opencv version: 4.1

acxz, 2019-06-13:
I ended up fixing the problem on my machine by changing the `OpenCV REQUIRED` in `image-view/CMakeLists.txt` to `OpenCV 3 REQUIRED`. It seems as though even if this package works with opencv I still need to patch it accordingly so I can use this package alongside others that require opencv3 specifically for now. Going ahead and closing this.

closed 2019-06-13 by acxz

SalilVishnuKapur, 2019-09-19:
@acxz Kindly guide me If I should manually fix all the obsolete opencv syntax to build it for opencv 4.1.0? I have no option left to build it with opencv 3. Issue : https://github.com/ros-perception/image_pipeline/issues/460

acxz, 2019-09-20:
@SalilVishnuKapur From what I understand this package should work with opencv4. See #375

SalilVishnuKapur, 2019-09-21:
Thanks for replying back and I will definitely check Arch linux branch. Actually, I was able to build image_pipeline with opencv 4.1.0. And now out of the ten ros packages just one is left, ar_track_alvar which is the last one, though I have debugged it but plan to do the fixes by monday. At 94% of building these ten ros packages. But now I want be part of this open source community, can you please add me as a maintainer to these ros packages? I am ready to take any kind responsibility for these open source ros packages.

acxz, 2019-09-21:
Arch Linux is not a branch it is another operating system like Debian or Ubuntu. Arch Linux is rolling release so it always has the most up to date upstream packages at any given time. For example, the default opencv on Arch is opencv4.

You can contribute to open source without being a maintainer. That is one of the beauties of open source, you don't have to "own" or officially be a part of the codebase to submit changes (pull requests (PRs)). Your enthusiasm is much appreciated. You should definitely share your changes on these packages by creating a "fork" of this repository, committing changes to it, and then submitting a PR upstream to the original codebase you forked from. If they are good changes then they would be merged into upstream for everyone to use.