ros-perception/image_pipeline #424: larger disparity_range for stereo_image_proc

mogumbo, 2019-07-01:
As far as I can tell, the disparity_range for stereo_image_proc is capped at 256 in Disparity.cfg and can only be changed by recompiling. Is there any other way to force a higher value? Is there any interest in permanently expanding the min/max values for min_disparity and disparity_range? It would be nice to eliminate the those limits completely, but I don't see a way to do that with the dynamic_reconfigure infrastructure that is in use.

SteveMacenski, 2019-07-10:
So [here](https://github.com/ros-perception/image_pipeline/blob/3b67763f8a692bdc2dfaec315cc8f027da236a91/stereo_image_proc/src/nodelets/disparity.cpp#L220) it shows that it just needs to be an increment of 16, not limited by size.

In the cfg file there's a [limit of 256](https://github.com/ros-perception/image_pipeline/blob/3b67763f8a692bdc2dfaec315cc8f027da236a91/stereo_image_proc/cfg/Disparity.cfg#L23) in dynamic reconfigure, but that's not a "real" limit, that's just a limit for the reconfigure system. No where in the processor does it strictly enforce that limit.

tl'dr: if you want a higher value, please submit a PR with a *valid and tested* new maximum cap for that cfg file and I'll merge it. You could also remove the limit, the minimum and maximum value fields are optional, I would merge that as well.

SteveMacenski, 2019-11-03:
Closing since just a dynamic reconfigure soft limit & no response

closed 2019-11-03 by SteveMacenski