ros-perception/image_pipeline #449: Rectify plethora of changes in ROS1 and ROS2 versions

SteveMacenski, 2019-09-03:
We've had alot of changes that are targeted at either ROS1 or ROS2 that have been changing quite alot, but nothing to homologate these changes together to make sure we have a consistent set of features and elements between ROS1 and ROS2.

mikeferguson, 2024-01-19:
List of PRs to review (some might actually be in ROS 2 - but this is everything I see that isn't clearly in ROS 2 when doing a diff between noetic and rolling)
 * [ ] #816 
 * [ ] #791 
 * [ ] #795 
 * [ ] #812 (may be needed?)
 * [ ] #806
 * [ ] #716 
 * [ ] #723
 * [ ] #671 
 * [ ] #630 
 * [ ] #489 
 * [ ] #375 
 * [ ] #336

mikeferguson, 2024-01-19:
https://github.com/ros-perception/image_pipeline/pull/863 ports #816 to ROS 2

ahcorde, 2024-01-19:
- Add support for floating point fps https://github.com/ros-perception/image_pipeline/pull/866 (rolling)

ahcorde, 2024-01-19:
- depth_image_proc/point_cloud_xyzi_radial Add intensity conversion (copy) for float https://github.com/ros-perception/image_pipeline/pull/867

ahcorde, 2024-01-19:
> @ahcorde - are you planning to do the last one on this list? (OpenCV4 compatibility - possible some of this is already in rolling, which makes it harder to actually resolve the diff)

I was checking this PR, Humble is supported from Ubuntu Focal and Debian Bullseye which I think we can only install OpenCV4 (not sure about OpenCV3). All the changes are already made, which means that these packages can compile with OpenCV 3 and 4

closed 2024-01-19 by mikeferguson