ros-perception/image_pipeline #451: image_proc/resize does not work with image_transport

shuntaraw, 2019-09-05:
`image_proc/resize` uses `image_transport` to subscribes to image and camera_info messages, but does not receive messages via `compressed` or `compressedDepth` transports. 

[repro]
1. Publish image and camera_info with `image_transport_plugins` such as `compressed_image_transport`:
```
$ rostopic list
/camera_info
/image_raw
/image_raw/compressed
:
```
2. Subscribe to  `/camera/image_raw/compressed` in `image_proc/resize` nodelet.
```
$ rosrun nodelet nodelet standalone image_proc/resize image:=/image_raw camera_info:=/camera_info _image_transport:=compressed
```

[expected]
`image` and `camera_info` are republished respectively at `/image_proc_resize/image` and `/image_proc_resize/camera_info`. Images are not resized because the default scale is 1. 

[actual]
`/image_proc_resize/image` and `/image_proc_resize/camera_info` are advertised but do not publish any message.

SteveMacenski, 2019-09-10:
Can you post an example message you’re sending over the compressed topic with the image data itself clipped?

mikeferguson, 2024-01-17:
Closing due to lack of response

closed 2024-01-17 by mikeferguson