ros-perception/image_pipeline #455: Build Failure [OpenCV 3]

acxz, 2019-09-14:
When attempting to build this package I obtain the following error:
```
-- Using CATKIN_DEVEL_PREFIX: /home/acxz/vcs/github/ros-melodic-arch/ros-melodic-image-view/src/build/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/melodic
-- This workspace overlays: /opt/ros/melodic
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.7.4", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using default Python package layout
-- Using empy: /usr/lib/python3.7/site-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Skip enable_testing() when building binary package
-- Using CATKIN_TEST_RESULTS_DIR: /home/acxz/vcs/github/ros-melodic-arch/ros-melodic-image-view/src/build/test_results
-- Found gtest: gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-3.7
-- catkin 0.7.17
-- BUILD_SHARED_LIBS is on
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Found PythonInterp: /usr/bin/python3 (found version "3.7.4") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/acxz/vcs/github/ros-melodic-arch/ros-melodic-image-view/src/build
make[2]: *** No rule to make target '/opt/opencv3/lib/libopencv_core.so.3.4.6', needed by 'devel/lib/image_view/disparity_view'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:146: CMakeFiles/disparity_view.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

It seems as tho when I upgraded my Opencv 3 to the latest version, 3.4.7, the package build broke for me. Can this be fixed via CMakeLists to allow later Opencv versions? or is it necessary for this package to remain on Opencv 3.4.6?

acxz, 2019-09-14:
Nvm a clean build fixed the issue.

closed 2019-09-14 by acxz

wbthomason, 2019-10-01:
@acxz Can you elaborate on what you did to get a working build? I have tried a clean build several times but keep encountering this issue.