ros-perception/image_pipeline #456: Build Failure (image_view) [Pango upgrade/harfbuzz]

acxz, 2019-09-14:
When trying to build image_view I am receiving the following error:
```
-- The C compiler identification is GNU 9.1.0
-- The CXX compiler identification is GNU 9.1.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/acxz/.cache/yay/ros-melodic-image-view/src/build/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/melodic
-- This workspace overlays: /opt/ros/melodic
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.7.4", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using default Python package layout
-- Found PY_em: /usr/lib/python3.7/site-packages/em.py  
-- Using empy: /usr/lib/python3.7/site-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Skip enable_testing() when building binary package
-- Using CATKIN_TEST_RESULTS_DIR: /home/acxz/.cache/yay/ros-melodic-image-view/src/build/test_results
-- Found gtest: gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-3.7
-- catkin 0.7.17
-- BUILD_SHARED_LIBS is on
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Found PythonInterp: /usr/bin/python3 (found version "3.7.4") 
-- Found Threads: TRUE  
-- Found Boost: /usr/include (found version "1.69.0") found components:  thread chrono date_time atomic 
-- Found OpenCV: /opt/opencv3 (found suitable version "3.4.7", minimum required is "3") 
-- Found GTK2_GTK: /usr/lib/libgtk-x11-2.0.so  
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    PYTHON_BASENAME
    PYTHON_INCLUDE_DIR
    PYTHON_LIBRARY

-- Build files have been written to: /home/acxz/.cache/yay/ros-melodic-image-view/src/build
Scanning dependencies of target disparity_view
[  5%] Building CXX object CMakeFiles/disparity_view.dir/src/nodes/disparity_view.cpp.o
[ 11%] Linking CXX executable devel/lib/image_view/disparity_view
[ 11%] Built target disparity_view
Scanning dependencies of target image_view
[ 17%] Building CXX object CMakeFiles/image_view.dir/src/nodelets/image_nodelet.cpp.o
[ 23%] Building CXX object CMakeFiles/image_view.dir/src/nodelets/disparity_nodelet.cpp.o
In file included from /usr/include/pango-1.0/pango/pango-font.h:25,
                 from /usr/include/pango-1.0/pango/pango-attributes.h:25,
                 from /usr/include/pango-1.0/pango/pango.h:25,
                 from /usr/include/gtk-2.0/gdk/gdktypes.h:37,
                 from /usr/include/gtk-2.0/gdk/gdkscreen.h:32,
                 from /usr/include/gtk-2.0/gdk/gdkapplaunchcontext.h:31,
                 from /usr/include/gtk-2.0/gdk/gdk.h:32,
                 from /usr/include/gtk-2.0/gtk/gtk.h:32,
                 from /home/acxz/.cache/yay/ros-melodic-image-view/src/image_pipeline-1.13.0/image_view/src/nodelets/disparity_nodelet.cpp:42:
/usr/include/pango-1.0/pango/pango-coverage.h:28:10: fatal error: hb.h: No such file or directory
   28 | #include <hb.h>
      |          ^~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/image_view.dir/build.make:76: CMakeFiles/image_view.dir/src/nodelets/disparity_nodelet.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:173: CMakeFiles/image_view.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```
As of right now the ArchLinux package is breaking because of this (https://github.com/ros-melodic-arch/ros-melodic-image-view/issues/6).

Will this package be updated to reflect changes in the pango release? This issue is also popping up in other software packages like: https://gitlab.kitware.com/cmake/cmake/issues/19531.

The fix for me was to add the following line before calling the `cmake` command:
```
export CXXFLAGS="${CXXFLAGS} -isystem /usr/include/harfbuzz"
```

Should the equivalent of this line be called within the CMakeLists.txt for this project?

ros-melodic-image-view ver: 1.13.0
pango ver: 1:1.44.6-1
gcc ver: 9.1
OS: Arch Linux

SteveMacenski, 2019-09-16:
PRs are welcome

acxz, 2019-09-16:
Yeah, harfbuzz headers could be installed in another location although I believe they are under the same location for Debian systems. This issue probably won't be seen for a while in Debian systems, since I think Ubuntu is still on pango ver: 1. If I find a better solution then I'll submit a PR, until then we have patched the ArchLinux package so that `ros-melodic-desktop-full` builds successfully.

PR #458 (2019-09-18): Provide include directory for harfbuzz
This resolves #456 . Changes to the latest pango release resulted in issues linking to harfbuzz headers. See: https://gitlab.gnome.org/GNOME/pango/issues/387.

Not sure if the solution presented in this PR is the best one, but it fixes the build. Here is a potential alternative solution: https://gitlab.com/Remmina/Remmina/commit/32adac791b8d112a8d618ef770326674092f5af7

See #456 for more information.
closed 2019-09-18 by JWhitleyAStuff