ros-perception/image_pipeline #460: error installing image_pipeline using catkin_make with OpenCV 4.1.0

SalilVishnuKapur, 2019-09-19:
Issue :
I have compiled opencv 4.1.0 with cuda support. So want to build the ros dependent packages with custom opencv. The command after initializing catkin_init_workspace is as follows. Any kind of help will be deeply appreciated
Command :
catkin_make -DCMAKE_BUILD_TYPE=Release -DOpenCV_DIR=/opt/include/opencv4/opencv2 -DPYTHON_EXECUTABLE=/usr/bin/python3

My fixes till yet :
I fixed the initial issues by editing the cmake file to have opencv version 4.1.0 and also other C++ and header files to have #include <opencv2/opencv.hpp> instead of #include <opencv/cv.h> etc.

My system config :
Ubuntu 18.04
Opencv 4.1.0
gcc 7.4.0
numpy 1.16.5
catkin

Log error(small snippet of the 1822 lines error log) :
/home/spiri/ros_opencv_dependables/src/image_pipeline/image_publisher/src/nodelet/image_publisher_nodelet.cpp: In member function ‘void image_publisher::ImagePublisherNodelet::do_work(const ros::TimerEvent&)’:
/home/spiri/ros_opencv_dependables/src/image_pipeline/image_publisher/src/nodelet/image_publisher_nodelet.cpp:98:20: error: ‘CV_CAP_PROP_POS_FRAMES’ was not declared in this scope
           cap_.set(CV_CAP_PROP_POS_FRAMES, 0);
                    ^~~~~~~~~~~~~~~~~~~~~~
/home/spiri/ros_opencv_dependables/src/image_pipeline/image_publisher/src/nodelet/image_publisher_nodelet.cpp: In member function ‘virtual void image_publisher::ImagePublisherNodelet::onInit()’:
/home/spiri/ros_opencv_dependables/src/image_pipeline/image_publisher/src/nodelet/image_publisher_nodelet.cpp:139:38: error: ‘CV_LOAD_IMAGE_COLOR’ was not declared in this scope
       image_ = cv::imread(filename_, CV_LOAD_IMAGE_COLOR);
                                      ^~~~~~~~~~~~~~~~~~~
/home/spiri/ros_opencv_dependables/src/image_pipeline/image_publisher/src/nodelet/image_publisher_nodelet.cpp:139:38: note: suggested alternative: ‘CV_IS_IMAGE_HDR’
       image_ = cv::imread(filename_, CV_LOAD_IMAGE_COLOR);
                                      ^~~~~~~~~~~~~~~~~~~
                                      CV_IS_IMAGE_HDR

SteveMacenski, 2019-09-19:
Unfortunately we don’t directly support openCV4. If you have code changes that fixes your problem and is backwards compatible, we’d be happy to review a PR.

closed 2019-09-19 by SteveMacenski

jorgemia, 2020-03-28:
@SalilVishnuKapur could you elaborate how you made it work with openCV 4 please? What files do I need to update?

BHamlen, 2020-04-14:
@jorgemia I had this same issue yesterday on an nvidia xavier (opencv 4.1.something same pre-installed) I fixed it by sudo apt purging opencv and building opencv 3.4 from source. That allowed image_pipeline to make succesfully

Well, almost. One other small thing with the making after I installed opencv3 was that I needed to change one of the lines in the makefile for image_pipeline, as its looking for a folder called opencv but the folder is actually called opencv4. Changing this in the make file allowed it to make succesfully. I dont remember exactly which file it is but just make and see where it fails, then go to that file.

Also also, if you're doing this on an nvidia jetson or other embedded platform, the built-in hard drive is not big enough, and the opencv build process actually fills it up, so you need to build it on a flash drive or something. I built it on a flash drive I plugged in, following this tutorialhttps://www.youtube.com/watch?v=08OAN9m0TjM