ros-perception/image_pipeline #461: No Status Bars

rukie, 2019-09-19:
I'm running ros-kinetic on Ubuntu 16 and attempting to run camera calibration on simulated data. 

I previously had issues with the GUI freezing on start-up because OpenCV 3.4 is on my machine. I decided to try building the package after installing through apt-get failed.
(Similar issue here: https://answers.ros.org/question/296087/camera_calibration-opens-gui-window-but-freezes/) 

After building locally with catkin_make, everything seems to function except for:
I don't have a status bar loading for x/y/skew. 

I was hoping not to uninstall OpenCV... thoughts on troubleshooting?

[image]

JWhitleyAStuff, 2019-09-19:
@rukie - I don't believe the bars will show up until the first successful measurement has been taken. What command are you using to run the calibration tool?

rukie, 2019-09-19:
The checkerboard is 14x14. I've found its recommended to search for something slightly smaller.
/unity_image/raw is republished by image_transport to convert from a compressed image, to a sensor/Image message.
Command used:
```bash
rosrun camera_calibration cameracalibrator.py --size 13x12 --square 0.05 image:=/unity_image/raw --no-service-check
```

rukie, 2019-09-19:
So it has to be 1 under in each direction. a 14x14 chessboard must be 13x13.
[image]

closed 2019-09-19 by rukie

JWhitleyAStuff, 2019-09-19:
That's because it is asking for the number of intersections, not the number of squares. It's a confusing parameter.