ros-perception/image_pipeline #476: Licensing Issues

JWhitleyAStuff, 2019-11-06:
All of the packages in this repo are currently licensed under BSD 3-clause. However, some developers require their contributions to be under Apache 2.0 to be allowed to contribute during regular work hours. Projects licensed under Apache 2.0 can contain code licensed under BSD but we need to figure out how to handle license notification and licensing of contributions going forward.

JWhitleyAStuff, 2019-11-06:
@tfoote, @dirk-thomas, or @wjwwood probably have experience with this. Hopefully they can offer some guidance.

SteveMacenski, 2019-11-06:
("some developers" are me, if that wasn't totally clear, so we don't need to keep this too abstract)

wjwwood, 2019-11-14:
I don't have an answer.

Though maybe we can dual license it. In that case I don't know what incoming contributions would be under (both or just one? who chooses?).

Properly changing the license to Apache from BSD is a huge pain though, since we have to seek permission from all past contributors, or at least that's my understanding.

wjwwood, 2019-11-14:
@gerkey do you have input on this?

SteveMacenski, 2019-11-14:
I'm not an expert on BSD <> Apache2.0 (and this question has come up a few times, I should talk to our lawyers), but off hand does anyone know if I can do `modifications made (C) Apache-2.0` on top of BSD? 

I'm sure I can get permissions to contribute with BSD, but it is a pain. Apache-2.0 I (and probably most companies) can just do it off hand without having to fill out paperwork.

gerkey, 2019-11-14:
The easiest and least confusing path from the point of view of maintaining and consuming the code is to contribute changes under the same license that was originally used (in this case, BSD-3).

If that's not possible, and we want the changes (and, as @wjwwood said, we don't want to try to collect permission from all previous contributors to change the original license), then we're adding an additional license to the code. That's not the end of the world.

Here's how I would approach it:
* new changes are being offered under a different, but compatible license (in this case, Apache-2);
* accepting those changes will make the code mixed-license, and the maintainer(s) of the repo need to be OK with that change;
* in each source file where differently licensed changes are introduced, the license header for the additional license (Apache-2) should be added, probably appended directly after the original license header (BSD-3); and
* in each package where such changes are made, the `package.xml` should be modified to list both licenses in the `<license>` tag (`BSD, Apache 2.0`).

closed 2019-11-25 by SteveMacenski