ros-perception/image_pipeline #483: Create Eloquent branch + CI

SteveMacenski, 2019-11-25:


JWhitleyAStuff, 2019-11-25:
The official ROS Docker Hub doesn't have `eloquent` containers yet. This will keep us from being able to do CI unless we want to build our own based on `ubuntu:bionic`, which would add a good amount of time to each build (10+ minutes). Up to you.

SteveMacenski, 2019-11-25:
The ROS build farm will have a PR builder for Eloquent that should be fine to start with

SteveMacenski, 2019-11-26:
@JWhitleyAStuff we also don't really have a dashing branch / CI, we have a master ros2 branch. Given Dashing is LTS we should have a separate branch for that now.

SteveMacenski, 2019-11-26:
Not sure if this is something you can work on, I'm currently dealing with PCL perception & RL

JWhitleyAStuff, 2019-11-26:
@SteveMacenski Do we want to wait until the whole stack is ported? I don't think we've actually done a release for Dashing yet, right?

SteveMacenski, 2019-11-26:
We haven't done a release yet, no, but also there were a number of changes in Eloquent, so we should really be keeping up with current or after we're done porting to ROS2, we're going to need to port from Dashing to G-turtle

JWhitleyAStuff, 2019-11-26:
Do you have Slack or a faster communication mechanism? I feel like we're using an issue as a chat board.

SteveMacenski, 2019-11-26:
I dont... There's also a ROS perception Riot (matrix client) Michael setup. 

+rosorg-perception:matrix.org

But I think this is also an appropriate discussion for a ticket

JWhitleyAStuff, 2019-11-26:
OK. So, how about we rename `ros2` to `dashing` and then branch for `eloquent`?

SteveMacenski, 2019-11-26:
Well we still want a master branch to work off of, I think ros2 meets that requirement.  

Usually how this has worked in the past is a master (ros2, whatever) branch to develop on, and before a release backport master to a distribution. 

I'm also fine not releasing to eloquent out of the gate, but Dashing should also be protected from master when we eventually go forward port to a newer distribution.

SteveMacenski, 2020-04-10:
@JWhitleyWork how do you feel about skipping Eloquent and moving to Foxy? 

Alternatively, we can see if we can get the ros2 master branch working on Eloquent as well and release it then. 

I think we should set an actual goal to start fresh in Foxy and have an actual foxy branch separate from ros2 & release debians.

caelinsutch, 2020-04-13:
To clarify is there any eloquent packages currently? Or will the ROS2 branch compile/run for eloquent? I'm looking for image_proc in eloquent.

SteveMacenski, 2020-04-13:
Go ahead and try with `ros2` and file a ticket if you have any issues. I suspect it will work. We're just now working through completing the initial ROS2 port so we haven't yet gotten to making the differentiation between specific distributions.

JWhitleyWork, 2020-04-14:
I think the `ros2` branch needs a hell of a lot of clean-up. Because of this, I think cleaning it up on `ros2` and then creating a Foxy branch/release is a more attainable goal by the time that Foxy is out than trying to clean up `ros2`, trying to make a working Eloquent release, then trying to make a Foxy release by the time Foxy is out. Thoughts?

SteveMacenski, 2020-04-14:
I don't understand what you just said. What I think we should do is do whatever is required on ros2 master and we can branch off for foxy when we're done.

closed 2020-06-01 by SteveMacenski