ros-perception/image_pipeline #496: PointCloud data is all Zeros/NaNs

nick-paul, 2020-01-17:
A point cloud is published with the correct transform but the XYZ data is entirely zeros and NaNs. 

[image]
*Rviz displays all points at the origin (visible axis is base_link). The console shows the output of a small c++ node I wrote which checks the point cloud coordinates. All values are zero (or NaN).*

[image]
*The params I am using for this test*

Below is some raw output from rostopic:

```
$ rostopic echo /brov2/camera/points2 | head -c 900
header: 
  seq: 30819
  stamp: 
    secs: 1579274882
    nsecs: 886668552
  frame_id: "cam_l"
height: 240
width: 320
fields: 
  - 
    name: "x"
    offset: 0
    datatype: 7
    count: 1
  - 
    name: "y"
    offset: 4
    datatype: 7
    count: 1
  - 
    name: "z"
    offset: 8
    datatype: 7
    count: 1
  - 
    name: "rgb"
    offset: 16
    datatype: 7
    count: 1
is_bigendian: False
point_step: 32
row_step: 10240
data: [0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 78, 63, 51, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 90, 71, 61, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 89, 71, 61, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 6, 26, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 192, 127, 0, 0, 1
```

## Other Notes

This issue appears to be related to #290 but it has since been closed for inactivity

These unanswered questions on answers.ros.org are describing the same issue and have additional detail:

  - https://answers.ros.org/question/333144/stereo_image_proc-does-not-generate-pointcloud/
  - https://answers.ros.org/question/321282/disparity-image-is-displayed-but-pointcloud-shows-only-a-small-dot-in-rviz/

This bag contains raw camera messages and the generated point clouds:
[points_and_cams.bag.zip](https://github.com/ros-perception/image_pipeline/files/3979506/points_and_cams.bag.zip)

Any idea what the problem might be?
Thanks

nick-paul, 2020-01-20:
Turns out the `camera_info` topics were not filled in correctly by the simulation I was using to generate the camera data. A few missing values caused all the matrix math to zero out and produce the above output. If anyone runs into this issue in the future, be sure to check that the data in your `camera_info` *([sensor_msgs/CameraInfo](https://docs.ros.org/melodic/api/sensor_msgs/html/msg/CameraInfo.html))* topics is correct. In my case the `P` matrix was not filled in completely.

It may be useful to add a warning message in the image callback if the `baseline` value is zero which would let users know their cameras are not configured correctly.

JWhitleyAStuff, 2020-01-20:
Thanks for posting the solution, @nick-paul ! Glad to hear it was fixed.

closed 2020-01-20 by JWhitleyAStuff

mfe7, 2020-05-22:
I had the same issue, where the realsense d435i `/camera/infra2/camera_info` used Tx=0 by default, which means baseline=0, which made my correct-looking disparity image result in a pointcloud where all points were at the origin. Updating the messages (I wrote a small new node) with P[3]=-Tx'*B fixes the issue and the pointcloud now looks right.

There seems to be some discussion in the realsense repo on whether this is right/wrong behavior: https://github.com/IntelRealSense/realsense-ros/pull/901

Petros626, 2022-05-20:
@nick-paul could I ask you just a simple question to the data: [...] you get?

nick-paul, 2022-05-24:
> could I ask you just a simple question to the data: [...] you get?

I'm not sure I understand your question; can you elaborate?

JMadhup, 2022-06-17:
@nick-paul my P matrix  seems fine but I still get no point cloud. Do you have any suggestion how I can deal with that?

Petros626, 2022-06-17:
> > could I ask you just a simple question to the data: [...] you get?
> 
> @Petros626 I'm not sure I understand your question; can you elaborate?

It's okay. It was about the 0 values in your data, I think it's from the manufacturer so produce 0 if object is out of range or something similar