ros-perception/image_pipeline #498: Image view seg faults on raw images

prasgane, 2020-02-12:
OS: Ubuntu 18.04
ROS Melodic

Image view seg faults when trying to run it from terminal. It seg faults only on raw images.

rosrun image_view image_view image:=camera/rgb/image_raw
[ INFO] [1581525926.314035143]: Initializing nodelet with 12 worker threads.
[ INFO] [1581525926.384639584]: Using transport "raw"
Segmentation fault (core dumped)

JWhitleyWork, 2020-02-12:
What type of camera are you using? What driver? Do you know what image format is being used for the `raw` stream?

prasgane, 2020-02-12:
I am using an Orbbec Astra pro with [this driver](https://github.com/orbbec/ros_astra_camera). It was working before I rand apt-get update and upgrade. 
I am not sure about the image format.

SteveMacenski, 2020-02-12:
Compile with debug symbols and get us a traceback when you have a moment.

prasgane, 2020-02-12:
I haven't build from source. I used apt-get to install it. The error persists when building from source too. Not sure if this is what you meant by traceback:
user@user:~/odom_comparision_ws$ cd ..
user@user:~$ mkdir -p image_ws/src
user@user:~$ cd image_ws/src/
user@user:~/image_ws/src$ git clone https://github.com/ros-perception/image_pipeline
Cloning into 'image_pipeline'...
remote: Enumerating objects: 50, done.
remote: Counting objects: 100% (50/50), done.
remote: Compressing objects: 100% (37/37), done.
remote: Total 8763 (delta 24), reused 22 (delta 12), pack-reused 8713
Receiving objects: 100% (8763/8763), 19.51 MiB | 808.00 KiB/s, done.
Resolving deltas: 100% (5417/5417), done.
user@user:~/image_ws/src$ cd ..
user@user:~/image_ws$ catkin_make -DCMAKE_BUILD_TYPE=RelWithDebInfo
Base path: /home/user/image_ws
Source space: /home/user/image_ws/src
Build space: /home/user/image_ws/build
Devel space: /home/user/image_ws/devel
Install space: /home/user/image_ws/install
Creating symlink "/home/user/image_ws/src/CMakeLists.txt" pointing to "/opt/ros/melodic/share/catkin/cmake/toplevel.cmake"
####
#### Running command: "cmake /home/user/image_ws/src -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCATKIN_DEVEL_PREFIX=/home/user/image_ws/devel -DCMAKE_INSTALL_PREFIX=/home/user/image_ws/install -G Unix Makefiles" in "/home/user/image_ws/build"
####
-- The C compiler identification is GNU 7.4.0
-- The CXX compiler identification is GNU 7.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/user/image_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/user/odom_comparision_ws/devel;/opt/ros/melodic
-- This workspace overlays: /home/user/odom_comparision_ws/devel;/opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/user/image_ws/build/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.20
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 8 packages in topological order:
-- ~~  - image_pipeline (metapackage)
-- ~~  - camera_calibration
-- ~~  - image_proc
-- ~~  - image_publisher
-- ~~  - image_view
[... 163 lines cut ...]
Scanning dependencies of target image_proc_exe
[ 86%] Building CXX object image_pipeline/image_proc/CMakeFiles/image_proc_exe.dir/src/nodes/image_proc.cpp.o
[ 88%] Building CXX object image_pipeline/stereo_image_proc/CMakeFiles/stereo_image_proc.dir/src/libstereo_image_proc/processor.cpp.o
[ 89%] Building CXX object image_pipeline/stereo_image_proc/CMakeFiles/stereo_image_proc.dir/src/nodelets/disparity.cpp.o
[ 91%] Building CXX object image_pipeline/stereo_image_proc/CMakeFiles/stereo_image_proc.dir/src/nodelets/point_cloud2.cpp.o
[ 93%] Linking CXX executable /home/user/image_ws/devel/lib/image_proc/image_proc
[ 93%] Built target image_proc_exe
[ 94%] Linking CXX shared library /home/user/image_ws/devel/lib/libdepth_image_proc.so
[ 94%] Built target depth_image_proc
[ 96%] Linking CXX shared library /home/user/image_ws/devel/lib/libstereo_image_proc.so
[ 96%] Built target stereo_image_proc
Scanning dependencies of target stereoimageproc_exe
[ 98%] Building CXX object image_pipeline/stereo_image_proc/CMakeFiles/stereoimageproc_exe.dir/src/nodes/stereo_image_proc.cpp.o
[100%] Linking CXX executable /home/user/image_ws/devel/lib/stereo_image_proc/stereo_image_proc
[100%] Built target stereoimageproc_exe
user@user:~/image_ws$ rosrun image_view image_view image:=camera/rgb/image_raw
[ INFO] [1581537758.203800698]: Initializing nodelet with 12 worker threads.
[ INFO] [1581537758.263336406]: Using transport "raw"
Segmentation fault (core dumped)

OS: Ubuntu 18.04
ROS Melodic
OpenCV (default that comes with ROS) 3.2

When I run debug with CLion, it faults on cvConvertImage() in  cv::imshow().

JWhitleyWork, 2020-02-12:
> OS: Ubuntu 18.04
> ROS Melodic
>
> Image view seg faults when trying to run it from terminal. It seg faults only on raw images.
>
> rosrun image_view image_view image:=camera/rgb/image_raw
> [ INFO] [...]

The topic you're using isn't a valid topic. It should start with `/`. Can you please run the `astra` driver again and give us the output of `rostopic list`?

SteveMacenski, 2020-02-12:
There is no requirement to start with `/` as far as I’m aware. 

Can you compile with gdb and get us a traceback, ei where its failing and why. 

If you think its from imshow, either your opencv install is incorrect or the image is invalid.

prasgane, 2020-02-12:
I am not so sure about the invalid topic because if it was the node wouldn't even subscribe to it and a rostopic echo work with that. I believe the / is to prevent any more namespaces from being added. 

I have been testing it off a bagfile I recorded a while back. The image_view node worked the same bag file until recent (if I recall right, last week) before I ran update and upgrade. But here is the rostopic list when I run the astra driver:

```
rostopic list 
/camera/camera_rgb/parameter_descriptions
/camera/camera_rgb/parameter_updates
/camera/rgb/camera_info
/camera/rgb/image_color
/camera/rgb/image_color/compressed
/camera/rgb/image_color/compressed/parameter_descriptions
/camera/rgb/image_color/compressed/parameter_updates
/camera/rgb/image_color/compressedDepth
/camera/rgb/image_color/compressedDepth/parameter_descriptions
/camera/rgb/image_color/compressedDepth/parameter_updates
/camera/rgb/image_color/theora
/camera/rgb/image_color/theora/parameter_descriptions
/camera/rgb/image_color/theora/parameter_updates
/camera/rgb/image_mono
/camera/rgb/image_mono/compressed
/camera/rgb/image_mono/compressed/parameter_descriptions
/camera/rgb/image_mono/compressed/parameter_updates
/camera/rgb/image_mono/compressedDepth
/camera/rgb/image_mono/compressedDepth/parameter_descriptions
/camera/rgb/image_mono/compressedDepth/parameter_updates
/camera/rgb/image_mono/theora
/camera/rgb/image_mono/theora/parameter_descriptions
/camera/rgb/image_mono/theora/parameter_updates
/camera/rgb/image_proc_debayer/parameter_descriptions
/camera/rgb/image_proc_debayer/parameter_updates
/camera/rgb/image_proc_rectify_color/parameter_descriptions
/camera/rgb/image_proc_rectify_color/parameter_updates
/camera/rgb/image_proc_rectify_mono/parameter_descriptions
/camera/rgb/image_proc_rectify_mono/parameter_updates
/camera/rgb/image_raw
/camera/rgb/image_raw/compressed
/camera/rgb/image_raw/compressed/parameter_descriptions
/camera/rgb/image_raw/compressed/parameter_updates
/camera/rgb/image_raw/compressedDepth
/camera/rgb/image_raw/compressedDepth/parameter_descriptions
/camera/rgb/image_raw/compressedDepth/parameter_updates
/camera/rgb/image_raw/theora
/camera/rgb/image_raw/theora/parameter_descriptions
/camera/rgb/image_raw/theora/parameter_updates
/camera/rgb/image_rect
/camera/rgb/image_rect/compressed
/camera/rgb/image_rect/compressed/parameter_descriptions
/camera/rgb/image_rect/compressed/parameter_updates
/camera/rgb/image_rect/compressedDepth
/camera/rgb/image_rect/compressedDepth/parameter_descriptions
/camera/rgb/image_rect/compressedDepth/parameter_updates
/camera/rgb/image_rect/theora
/camera/rgb/image_rect/theora/parameter_descriptions
/camera/rgb/image_rect/theora/parameter_updates
/camera/rgb/image_rect_color
/camera/rgb/image_rect_color/compressed
/camera/rgb/image_rect_color/compressed/parameter_descriptions
/camera/rgb/image_rect_color/compressed/parameter_updates
/camera/rgb/image_rect_color/compressedDepth
/camera/rgb/image_rect_color/compressedDepth/parameter_descriptions
/camera/rgb/image_rect_color/compressedDepth/parameter_updates
/camera/rgb/image_rect_color/theora
/camera/rgb/image_rect_color/theora/parameter_descriptions
/camera/rgb/image_rect_color/theora/parameter_updates
/clock
/rosout
/rosout_agg
/tf_static

```
image_view node works fine on the /camera/depth_registered/image_raw and /camera/rgb/image_mono topics.

JWhitleyWork, 2020-02-12:
Yeah, sorry about the `/` comment. Is the bag something we could download for testing?

prasgane, 2020-02-12:
@SteveMacenski is this helpful?

```
rosrun --prefix 'gdb -ex run --args' image_view image_view 
GNU gdb (Ubuntu 8.1-0ubuntu3.2) 8.1.0.20180409-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /opt/ros/melodic/lib/image_view/image_view...(no debugging symbols found)...done.
Starting program: /opt/ros/melodic/lib/image_view/image_view 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffee8a6700 (LWP 7688)]
[New Thread 0x7fffee0a5700 (LWP 7689)]
[New Thread 0x7fffed8a4700 (LWP 7690)]
[New Thread 0x7fffed0a3700 (LWP 7691)]
[New Thread 0x7fffec8a2700 (LWP 7692)]
[New Thread 0x7fffe7fff700 (LWP 7693)]
[New Thread 0x7fffe77fe700 (LWP 7694)]
[New Thread 0x7fffe6ffd700 (LWP 7695)]
[New Thread 0x7fffe67fc700 (LWP 7696)]
[New Thread 0x7fffe5ffb700 (LWP 7697)]
[New Thread 0x7fffe57fa700 (LWP 7698)]
[New Thread 0x7fffe4ff9700 (LWP 7699)]
[New Thread 0x7fffdffff700 (LWP 7700)]
[New Thread 0x7fffdf7fe700 (LWP 7701)]
[New Thread 0x7fffdeffd700 (LWP 7702)]
[New Thread 0x7fffde7fc700 (LWP 7703)]
[New Thread 0x7fffddffb700 (LWP 7706)]
[ INFO] [1581540770.534065584]: Initializing nodelet with 12 worker threads.
[New Thread 0x7fffdd7fa700 (LWP 7707)]
[ INFO] [1581540771.270263212, 1579908324.628823872]: Using transport "raw"
[New Thread 0x7fffb5664700 (LWP 7708)]
[New Thread 0x7fffafdc2700 (LWP 7709)]
[New Thread 0x7fffaf5c1700 (LWP 7710)]

Thread 20 "image_view" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffb5664700 (LWP 7708)]
0x00007fffe4169220 in cvConvertImage () from /usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.3.2
```
I am really surprised about the cv errors considering that it works for other types or images (mono and depth) and dont think the images are incorrect considering that the bag file worked few days back and this error happens when I connect the camera and run it live.  I have been able to repeat this in another computer in our lab (with the same setup 18.04 and Melodic) which hadnt been updated. The node break after I run update and upgrade so something in the update might have broken it?

@JWhitleyWork the bag file is not publicly available but I can share a link over DM/email.

JWhitleyWork, 2020-02-12:
One more piece of information would help: Can you `rostopic echo /camera/rgb/image_raw | grep encoding` and let us know what encoding it is being published as?

SteveMacenski, 2020-02-12:
So you're 90% there, after it crashes in the gdb sessions type "backtrace" and that's what we really need. That will give you a full rundown of what was called and how

prasgane, 2020-02-12:
encoding: "bgr8"

prasgane, 2020-02-12:
```
(gdb) backtrace
#0  0x00007fffe4169220 in cvConvertImage ()
    at /usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.3.2
#1  0x00007fffdc73a897 in  () at /usr/lib/x86_64-linux-gnu/libopencv_highgui.so.3.2
#2  0x00007fffdc73cd3c in cvShowImage () at /usr/lib/x86_64-linux-gnu/libopencv_highgui.so.3.2
#3  0x00007fffdc739734 in cv::imshow(cv::String const&, cv::_InputArray const&) ()
    at /usr/lib/x86_64-linux-gnu/libopencv_highgui.so.3.2
#4  0x00007fffe45d7b57 in image_view::ImageNodelet::windowThread() ()
    at /opt/ros/melodic/lib//libimage_view.so
#5  0x00007ffff5b53bcd in  () at /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.65.1
#6  0x00007ffff592a6db in start_thread (arg=0x7fffb5664700) at pthread_create.c:463
#7  0x00007ffff6d5a88f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

SteveMacenski, 2020-02-12:
This appears to be from your `/opt/ros/melodic` install not your workspace build

prasgane, 2020-02-12:
```
(gdb) backtrace 
#0  0x00007fffe4169220 in cvConvertImage () at /usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.3.2
#1  0x00007fffd873a897 in  () at /usr/lib/x86_64-linux-gnu/libopencv_highgui.so.3.2
#2  0x00007fffd873cd3c in cvShowImage () at /usr/lib/x86_64-linux-gnu/libopencv_highgui.so.3.2
#3  0x00007fffd8739734 in cv::imshow(cv::String const&, cv::_InputArray const&) ()
    at /usr/lib/x86_64-linux-gnu/libopencv_highgui.so.3.2
#4  0x00007fffe45d7b57 in image_view::ImageNodelet::windowThread() (this=0x5555560619b0)
    at /home/user/image_ws/src/image_pipeline/image_view/src/nodelets/image_nodelet.cpp:234
#5  0x00007ffff5b53bcd in  () at /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.65.1
#6  0x00007ffff592a6db in start_thread (arg=0x7fffb56b5700) at pthread_create.c:463
#7  0x00007ffff6d5a88f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```
Sorry, here is the one from my workspace build.

b4795, 2020-02-13:
I also have this issue. But a few days ago I did not have this problem.

Could this caused by the recent update from 1.13.0-1bionic.20191211.032606 to 1.14.0-1bionic.20200116.135419 (Ubuntu 18.04)?

In my "/var/log/apt/history.log" I see the following (edited for clarity):
```
Start-Date: 2020-02-10  13:30:34
Commandline: aptdaemon role='role-commit-packages' sender=':1.130'
Upgrade: [...], ros-melodic-image-view:amd64 (1.13.0-1bionic.20191211.032606, 1.14.0-1bionic.20200116.135419), [...]
End-Date: 2020-02-10  13:31:40
```

SteveMacenski, 2020-02-13:
@JWhitleyWork it looks like you were the last to touch this file - any input?

Like Josh said, can you provide us with a _short_ bag file to start from to make sure we're all on the same page?

prasgane, 2020-02-13:
I have emailed you a link containing a bag file.

SteveMacenski, 2020-04-10:
Pinging @JWhitleyWork  - it looks like you may have broken this from my Feb 13 comment

closed 2020-05-06 by SteveMacenski

Shillin123, 2023-02-15:
Hello, did you solve this? I have been stuck at this problem for several days. I use Noetic, when I rosrun image_view image_view image:=/panda/camera/image_raw, I get segment fault. I try to record a rosbag and check the info, I can get this:
path:        2023-02-15-15-18-26.bag
version:     2.0
duration:    7:47s (467s)
start:       Jan 01 1970 01:08:38.38 (518.38)
end:         Jan 01 1970 01:16:25.72 (985.72)
size:        6.3 GB
messages:    14022
compression: none [7011/7011 chunks]
types:       sensor_msgs/Image [060021388200f6f0f447d0fcd9c64743]
topics:      /panda/camera/image_raw   14022 msgs    : sensor_msgs/Image
And then I play the bagfile, I can receive the bag time  and duration. And when I run rqt_image_view, it's good as well and I can see a new window with image. So it seems no problem with the node, but it just dies always when I try to rosrun image_view image_view image:=/panda/camera/image_raw. Can you help me with this?