ros-perception/image_pipeline #505: monodepth usage

amrehab410, 2020-03-19:
I was wondering if the depth_image_proc package can be used with depth images generated from mono images?

valgur, 2020-03-20:
You can do that sure, but `depth_image_proc` itself won't help you too much. You can create a Python node with an `Image` subscriber and `DisparityImage` publisher that applies depth prediction to the input image.
I have written such wrapper node at some point that I could share, but I can't find it right now. For `monodepth2`, you might find my fork that restructures it into a setup.py-installable library useful: https://github.com/valgur/monodepth2/blob/master/monodepth2/__init__.py

amrehab410, 2020-03-20:
Looks like I didn't explain the inquiry in a correct way, I need to generate a pointcloud2 msg from a depth image(generated from monodepth2), I've tried but there isn't any output from /points topic

SteveMacenski, 2020-04-10:
Please ask on ROS answers, thanks!

closed 2020-04-10 by SteveMacenski