ros-perception/image_pipeline #506: Buffer overrun with message length [0]

seanrivera, 2020-03-23:
Hello, 

I am performing a fault sensitivity analysis on several common ROS nodes and discovered that there is a buffer overrun in the image_proc node. It seems like the proc node simply works with the deserialized image after execution. 

Exception thrown when deserializing message of length [0] from [/test_node]: Buffer Overrun

SteveMacenski, 2020-03-23:
PRs welcome!

SteveMacenski, 2020-04-10:
Hi, 

Any update on this @seanrivera ?

seanrivera, 2020-04-13:
Hi, 

Sorry about the delay in getting back. I haven't had time to dive into the issue for the past couple weeks.

SteveMacenski, 2020-05-20:
@seanrivera any time to work on this? Or provide more information so someone can replicate / fix?

mikeferguson, 2024-01-19:
No updates in years - and not enough information to actually be actionable

closed 2024-01-19 by mikeferguson