ros-perception/image_pipeline #515: ROS2 image_proc not creating any topics

caelinsutch, 2020-04-13:
My Launch File: 

```
import launch
import os
from ament_index_python.packages import get_package_share_directory
from launch_ros.actions import ComposableNodeContainer
from launch_ros.descriptions import ComposableNode
from launch_ros.actions import Node

def generate_launch_description():

    # Takes in gst-launch from gscam.py, which is a t for Rocos and a UDP src for ROS
    udpConfig = 'udpsrc port=5000 ! application/x-rtp,encoding-name=JPEG,payload=26 ! rtpjpegdepay ! jpegdec ! videoconvert'
    # Takes in the straight src from sensor 1
    vanillaConfig = "nvarguscamerasrc sensor-id=1 ! 'video/x-raw(memory:NVMM),width=1280,height=720,framerate=15/1, format=NV12, framerate=15/1' ! nvvidconv flip-method=2 ! video/x-raw, format=BGRx ! videoconvert"
    config_dir = os.path.join(get_package_share_directory('br-perception'), 'config')

    rightCameraConfig = 'file://' + os.path.join(config_dir, "ost_right.yaml")
    leftCameraConfig = 'file://' + os.path.join(config_dir, "ost_left.yaml")

    rightComposableNode = ComposableNode(
        package='gscam',
        node_plugin='gscam::GSCamNode',
        node_name='gscam_right',
        remappings=[
            ('/image_raw', '/stereo/right/image_raw'),
            ('/camera_info', '/stereo/right/camera_info')
        ],
        parameters=[
            os.path.join(get_package_share_directory('br-perception'), "config", "params.yaml"),
            {'camera_name': 'right'},
            {'gscam_config': udpConfig},
            {'camera_info_url': rightCameraConfig},
        ],
        extra_arguments=[{'use_intra_process_comms': True}],
    )
    leftComposableNode = ComposableNode(
        package='gscam',
        node_plugin='gscam::GSCamNode',
        node_name='gscam_left',
        remappings=[
            ('/image_raw', '/stereo/left/image_raw'),
            ('/camera_info', '/stereo/right/camera_info')
        ],
        parameters=[
            os.path.join(get_package_share_directory('br-perception'), "config", "params.yaml"),
            {'camera_name': 'left'},
            {'gscam_config': vanillaConfig},
            {'camera_info_url': leftCameraConfig},
        ],
        extra_arguments=[{'use_intra_process_comms': True}],
    )

    leftImageRect = ComposableNodeContainer(
        package='rclcpp_components',
        node_executable='component_container',
        node_name='image_proc_container',
        node_namespace='',
        composable_node_descriptions=[
            ComposableNode(
                package='image_proc',
                node_plugin='image_proc::RectifyNode',
                node_name='rectify_color',
                node_namespace='/stereo/left',
                parameters=[
                    {'camera_namespace': '/stereo/left'},
                    {'image_color': '/image_raw'}
                ]
            ),
        ],
    )

    gstreamerNode = Node(
        package='br-perception',
        node_executable='gscam.py',
        node_name='gscam_udp'
    )

    composableNodeContainer = ComposableNodeContainer(
        node_name='stereo',
        node_namespace='',
        package='rclcpp_components',
        node_executable='component_container',
        composable_node_descriptions=[
            rightComposableNode,
            # leftComposableNode
        ],
        output='screen',
    )

    return launch.LaunchDescription([composableNodeContainer, leftImageRect, gstreamerNode])
```

Excuse the poorly written code :)

Everything launches fine, the image_proc node runs, but no topics are outputed.

JWhitleyWork, 2020-04-13:
Would you mind providing us with the following information?

OS Version:

ROS Version:

`image_proc` version or commit hash:

When running, output of `ros2 node info /rectify_color:

caelinsutch, 2020-04-15:
Update - I submitted a pull request with the fixes made and I've attached the working launch file below

```
import launch
import os
from ament_index_python.packages import get_package_share_directory
from launch_ros.actions import ComposableNodeContainer
from launch_ros.descriptions import ComposableNode
from launch_ros.actions import Node

def generate_launch_description():

    # Takes in gst-launch from gscam.py, which is a t for Rocos and a UDP src for ROS
    # rightConfig='nnvarguscamerasrc sensor-id=1 ! video/x-raw(memory:NVMM), width=256, height=144, format=NV12, framerate=15/1 ! nvvidconv flip-method=2 ! tee name=t1 t1. ! nvideo/x-raw, format=BGRx ! videoconvert t1. ! nvv4l2h264enc bitrate=8000000 ! h264parse config-interval=1 ! queue silent=true leaky=downstream ! rndbuffersize max=65000 ! udpsink host=127.0.0.1 port=5001 sync=false '
    rightConfig = 'udpsrc port=5000 ! application/x-rtp,encoding-name=JPEG,payload=26 ! rtpjpegdepay ! jpegdec ! videoconvert'
    # Takes in the straight src from sensor 1
    # rightConfig = 'nvarguscamerasrc sensor-id=0 ! video/x-raw(memory:NVMM), width=1280, height=720, format=NV12, framerate=15/1 ! nvvidconv flip-method=2 ! video/x-raw, format=BGRx ! videoconvert'
    leftConfig = 'nvarguscamerasrc sensor-id=1 ! video/x-raw(memory:NVMM), width=256, height=144, format=NV12, framerate=15/1 ! nvvidconv flip-method=2 ! video/x-raw, format=BGRx ! videoconvert'
    config_dir = os.path.join(get_package_share_directory('br-perception'), 'config')

    rightCameraConfig = 'file://' + os.path.join(config_dir, "ost_right.yaml")
    leftCameraConfig = 'file://' + os.path.join(config_dir, "ost_left.yaml")

    rightNamespace = "/stereo/right"
    leftNamespace = "/stereo/left"

    rightGscam = ComposableNode(
        package='gscam',
        node_plugin='gscam::GSCamNode',
        node_name='gscam',
        node_namespace= rightNamespace,
        remappings=[
            ('/image_raw', '/stereo/right/image_raw'),
            ('/camera_info', '/stereo/right/camera_info')
        ],
        parameters=[
            os.path.join(get_package_share_directory('br-perception'), "config", "params.yaml"),
            {'camera_name': 'right'},
            {'gscam_config': rightConfig},
            {'camera_info_url': rightCameraConfig},
        ],
        extra_arguments=[{'use_intra_process_comms': True}],
    )
    leftGscam = ComposableNode(
        package='gscam',
        node_plugin='gscam::GSCamNode',
        node_name='gscam',
        node_namespace=leftNamespace,
        remappings=[
            ('/image_raw', '/stereo/left/image_raw'),
            ('/camera_info', '/stereo/right/camera_info')
        ],
        parameters=[
            os.path.join(get_package_share_directory('br-perception'), "config", "params.yaml"),
            {'camera_name': 'left'},
            {'gscam_config': leftConfig},
            {'camera_info_url': leftCameraConfig},
        ],
        extra_arguments=[{'use_intra_process_comms': True}],
    )

    rightImageRect = ComposableNode(
        package='image_proc',
        node_plugin='image_proc::RectifyNode',
        node_name='rectify_color',
        node_namespace=rightNamespace,
        parameters=[
            {"camera_namespace": rightNamespace},
            {"image_color": "/image_raw"}
        ]
    )

    gstreamerNode = Node(
        package='br-perception',
        node_executable='gscam.py',
        node_name='gscam_udp'
    )

    composableNodeContainer = ComposableNodeContainer(
        node_name='stereo',
        node_namespace='',
        package='rclcpp_components',
        node_executable='component_container',
        composable_node_descriptions=[
            rightGscam,
            leftGscam,
            rightImageRect
        ],
        output='screen',
    )

    return launch.LaunchDescription([composableNodeContainer, gstreamerNode])
```

SteveMacenski, 2020-04-15:
@caelinsutch what updates did you actually need to make? It seems like updating the launch file. I see a ton of work that is undescribed in the code itself - can you clarify what you did and why?

caelinsutch, 2020-04-15:
@SteveMacenski I took some edits from #457. A lot of the changes seem to be reformatting issues, so I may have to fix that. As I moved from CV3 to CV4, I added namespace on some of the enums to ensure that they would compile properly

SteveMacenski, 2020-04-15:
OK, lets separate these into 2 tickets. This one is for image proc launch file / not working. Lets do that in a PR and lets have a separate opencv4 PR.

My guess is a bunch of your changes are unnecessary or formatting issues. Please make use of the linters and only change lines you actually need to touch.

caelinsutch, 2020-04-15:
@SteveMacenski I made a branch for CV4 changes [here](https://github.com/BytesRobotics/image_pipeline/tree/ros2-cv4), how would you like me to open a PR?

SteveMacenski, 2020-04-15:
@caelinsutch does your new launch file work (vs old one not)? If so, please submit just that in a PR and we can merge it probably immediately. 

After that's in, then that means as far as opencv3 goes, I think that makes everything work. At that point, I can branch off a dashing/eloquent branch with those changes and then we can work on integrating the opencv4 into the `ros2` main branch. Alternatively, it may be beneficial to support both opencv 3 and 4 in one branch. It looks like you only had to make about 3 lines of changes. Maybe we can use ifdefs like you removed here (https://github.com/ros-perception/image_pipeline/compare/ros2...BytesRobotics:ros2-cv4#diff-dc51bb3665e45a9f6b0ebb51bb5a403bR49) for supporting them both.

caelinsutch, 2020-04-15:
@SteveMacenski I updated just the launch file (here)[https://github.com/BytesRobotics/image_pipeline/tree/ros2] with a working version

SteveMacenski, 2020-04-15:
Please submit a PR, Josh and I will review and merge. I'm happy as is with that but it looks like you may be missing some remaps:

                        ('image_mono', 'image_color'),
                        ('image_rect', 'image_rect_color')

caelinsutch, 2020-04-15:
Update, after some more testing it's not ready to go, I'll submit a PR tomorrow. Let me know if theres anything you want changed in the CV4 version

SteveMacenski, 2020-04-16:
Ok, @caelinsutch looking forward to it

SteveMacenski, 2020-05-20:
I think we've merged this in

closed 2020-05-20 by SteveMacenski