ros-perception/image_pipeline #516: image_view not functioning (segmentation fault)

BHamlen, 2020-04-14:
I'm running ros melodic on an nvidia xavier agx, ubuntu 18.04

Running `roslaunch video_stream_opencv video_file.launch` fails with the following output

```
... logging to /home/sd/.ros/log/d688c298-7de0-11ea-b5c8-00044be59e11/roslaunch-sd-17072.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://sd:38003/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.5
 * /videofile/videofile_stream/buffer_queue_size: 1000
 * /videofile/videofile_stream/camera_info_url: 
 * /videofile/videofile_stream/camera_name: videofile
 * /videofile/videofile_stream/flip_horizontal: False
 * /videofile/videofile_stream/flip_vertical: False
 * /videofile/videofile_stream/fps: 30.0
 * /videofile/videofile_stream/frame_id: videofile_frame
 * /videofile/videofile_stream/height: 0
 * /videofile/videofile_stream/loop_videofile: True
 * /videofile/videofile_stream/set_camera_fps: 30.0
 * /videofile/videofile_stream/start_frame: 0
 * /videofile/videofile_stream/stop_frame: -1
 * /videofile/videofile_stream/video_stream_provider: /home/sd/Desktop/...
 * /videofile/videofile_stream/width: 0

NODES
  /videofile/
    videofile_image_view (image_view/image_view)
    videofile_stream (video_stream_opencv/video_stream)

auto-starting new master
process[master]: started with pid [17082]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to d688c298-7de0-11ea-b5c8-00044be59e11
process[rosout-1]: started with pid [17093]
started core service [/rosout]
process[videofile/videofile_stream-2]: started with pid [17100]
process[videofile/videofile_image_view-3]: started with pid [17101]
[ INFO] [1586821523.337109716]: Initializing nodelet with 8 worker threads.
[ INFO] [1586821523.696650797]: Camera name: videofile
[ INFO] [1586821523.696753777]: Provided camera_info_url: ''
[ INFO] [1586821523.696830356]: Publishing with frame_id: videofile_frame
[ INFO] [1586821523.696888630]: Setting camera FPS to: 30
[ INFO] [1586821523.696934808]: Throttling to fps: 30
[ INFO] [1586821523.696980026]: Setting buffer size for capturing frames to: 1000
[ INFO] [1586821523.697021563]: Flip horizontal image is: false
[ INFO] [1586821523.697094270]: Flip vertical image is: false
[ INFO] [1586821523.697139808]: Video start frame is: 0
[ INFO] [1586821523.697176929]: Video stop frame is: -1
[ INFO] [1586821523.983021552]: using default calibration URL
[ INFO] [1586821523.983229944]: camera calibration URL: file:///home/sd/.ros/camera_info/videofile.yaml
[ INFO] [1586821523.983430912]: Unable to open camera calibration file [/home/sd/.ros/camera_info/videofile.yaml]
[ WARN] [1586821523.983517764]: Camera calibration file /home/sd/.ros/camera_info/videofile.yaml not found.
[ INFO] [1586821523.983828432]: Opening VideoCapture with provider: /home/sd/Desktop/video_stream_test.mp4
[ INFO] [1586821524.092015541]: Video stream provider type detected: videofile
[ INFO] [1586821524.092263775]: Camera reports FPS: 29.97
[ WARN] [1586821524.165497689]: No calibration file given, publishing a reasonable default camera info.
[ INFO] [1586821524.165714145]: The image width is: 1920
[ INFO] [1586821524.165816869]: The image height is: 1080
[videofile/videofile_image_view-3] process has died [pid 17101, exit code -11, cmd /opt/ros/melodic/lib/image_view/image_view image:=image_raw __name:=videofile_image_view __log:=/home/sd/.ros/log/d688c298-7de0-11ea-b5c8-00044be59e11/videofile-videofile_image_view-3.log].
log file: /home/sd/.ros/log/d688c298-7de0-11ea-b5c8-00044be59e11/videofile-videofile_image_view-3*.log 
[... 135 lines cut ...]
[roslaunch.pmon][INFO] 2020-04-13 18:45:30,022: ProcessMonitor exit: killing rosout-1
[roslaunch][INFO] 2020-04-13 18:45:30,022: [rosout-1] killing on exit
[roslaunch][INFO] 2020-04-13 18:45:30,022: process[rosout-1]: killing os process with pid[17093] pgid[17093]
[roslaunch][INFO] 2020-04-13 18:45:30,023: [rosout-1] sending SIGINT to pgid [17093]
[roslaunch][INFO] 2020-04-13 18:45:30,023: [rosout-1] sent SIGINT to pgid [17093]
[roslaunch][INFO] 2020-04-13 18:45:30,224: process[rosout-1]: SIGINT killed with return value 0
[roslaunch.pmon][INFO] 2020-04-13 18:45:30,224: ProcessMonitor exit: killing master
[roslaunch][INFO] 2020-04-13 18:45:30,224: [master] killing on exit
[roslaunch][INFO] 2020-04-13 18:45:30,225: process[master]: killing os process with pid[17082] pgid[17082]
[roslaunch][INFO] 2020-04-13 18:45:30,225: [master] sending SIGINT to pgid [17082]
[roslaunch][INFO] 2020-04-13 18:45:30,225: [master] sent SIGINT to pgid [17082]
[roslaunch][INFO] 2020-04-13 18:45:30,426: process[master]: SIGINT killed with return value 0
[roslaunch.pmon][INFO] 2020-04-13 18:45:30,426: ProcessMonitor exit: cleaning up data structures and signals
[roslaunch.pmon][INFO] 2020-04-13 18:45:30,427: ProcessMonitor exit: pmon has shutdown
[roslaunch][INFO] 2020-04-13 18:45:30,436: process monitor is done spinning, initiating full shutdown
[roslaunch][INFO] 2020-04-13 18:45:30,436: runner.stop()
[roslaunch][INFO] 2020-04-13 18:45:30,437: shutting down processing monitor...
[roslaunch][INFO] 2020-04-13 18:45:30,437: shutting down processing monitor <ProcessMonitor(ProcessMonitor-1, stopped daemon 548115911152)>
[roslaunch.pmon][INFO] 2020-04-13 18:45:30,437: ProcessMonitor.shutdown <ProcessMonitor(ProcessMonitor-1, stopped daemon 548115911152)>
[roslaunch][INFO] 2020-04-13 18:45:30,437: ... shutting down processing monitor complete
[roslaunch][INFO] 2020-04-13 18:45:30,437: done
[roslaunch.pmon][INFO] 2020-04-13 18:45:30,438: ProcessMonitor.shutdown <ProcessMonitor(ProcessMonitor-1, stopped daemon 548115911152)>
[rospy.core][INFO] 2020-04-13 18:45:30,438: signal_shutdown [atexit]

```

SteveMacenski, 2020-04-14:
I'm confused, https://wiki.ros.org/video_stream_opencv is not in this repo. Did you mean to file a ticket there?

BHamlen, 2020-04-14:
@SteveMacenski Ah, sorry for the confusion. I am almost certain the issue lies with image_view. Just wanted to provide context of where the image topic i'm feeding image_view is coming from

BHamlen, 2020-04-17:
@SteveMacenski do you have any updates on image_view's issues? I saw in a different thread that it was possible some dev was the last person to push changes to it and might have messed it up? Just to reiterate, this issue thread is about image_view. Sorry for the earlier confusion

SteveMacenski, 2020-04-17:
Can you reproduce using a much simpler demo? You have alot of stuff going on in your description, boiling it down to a simple issue with just this tool. 

Also verify as you mention that you’re compiling against the right version of opencv.

aytimothy, 2020-05-03:
It could be something to do with the environment that we're running in? I started from a Ubuntu Server setup and installed ROS, a desktop and all that.

aytimothy, 2020-05-03:
I have a crash dump here: https://aytimothy.xyz/files/pub/_opt_ros_melodic_lib_image_view_image_view.1001.crash

awesomebytes, 2020-05-03:
@aytimothy Have you tried recording a short rosbag of the `image_raw` `camera_info` and maybe `image_raw/compressed` topics and then trying to run `image_view` playing this rosbag? If it crashes that way, it's a very easy way for developers to reproduce the issue and fix it. (And the rosbag can be attached to the comment of the issue so it doesn't depend on external hosts).

aytimothy, 2020-05-04:
Okay, there's no point in posting a rosbag, because there's literally nothing to report (the rosbag file is empty), besides it saying "topic is not used" in the console even after started.

But what seems to be happening is that after it dies the first time, `roslaunch` starts it up again a few seconds later, and everything works fine. So, for this second launch, I doubt it's related to this. I think that crash dump file might be more useful than grabbing a `rosbag`.

JWhitleyWork, 2020-05-06:
@aytimothy Can you please test to see if the issue is resolved now that #526 is merged into Melodic?

awesomebytes, 2020-05-06:
@aytimothy Can you try to record a rosbag with having `visualize=false` (so some image from the `video_stream_opencv` driver gets published) and try to start `image_view` with the rosbag playing to see if it crashes? Just to confirm that it isn't related somehow to the format of the image published. If it would crash `image_view` it would be a perfect test case.

> But what seems to be happening is that after it dies the first time, roslaunch (or something) starts it up again a few seconds later, and everything works fine.

I don't know what may cause that behaviour. The launchfiles from `video_stream_opencv` don't contain a `respawn=true` value. There is also an optional parameter `reopen_on_read_failure` on the dynamic reconfigure, but it's also `False` by default. That's all I can think that may do something of that nature.

I admit that if this issue has to do with some kind of timing/racing condition issue of a driver starting to advertise images and `image_view` trying to connect/callback/some_other_behaviour the rosbag will be useless. But if it's related in some way with the type of data contained in the image (which both reports come from `video_stream_opencv` so they are generated with the same code) it should be reproducible with a rosbag. Maybe the first image published has some issue? Dunno, just trying to help.

JWhitleyWork, 2020-05-06:
@awesomebytes The `visualize` parameter doesn't exist in `image_view` and it never should have, IMO. The `image_view` nodelet is not intended as an image processor - there are plenty of other nodelets in this stack for that purpose - it is intended _only_ as an "image viewer." It should produce a GUI window which displays images from a given topic and that should be it's only purpose. If some minor image processing is necessary to achieve that goal then it can be done internally but the images that are processed therein _should not_ be published to a topic lest it be misused as an "image processing" node.

awesomebytes, 2020-05-07:
@JWhitleyWork the `visualize` parameter exists in the launchfiles from `video_stream_opencv`. As I believe we were having some kind of communication issue I was more detailed with the explanation of how and why to record a rosbag that may contain data that may trigger the crash in `image_view`.

JWhitleyWork, 2020-05-07:
@awesomebytes Thanks for the clarification.

hoonkai, 2020-05-17:
> Have you tried recording a short rosbag of the `image_raw` `camera_info` and maybe `image_raw/compressed` topics and then trying to run `image_view` playing this rosbag? If it crashes that way, it's [...]

@awesomebytes Hi I'm having the same issue. I've tried recording a rosbag as per your advice, but I get an error after 200 msgs:

> [ERROR] [1589731616.096380328]: Could not capture frame
> [ERROR] [1589731616.129171038]: Could not capture frame
> [ERROR] [1589731616.162531012]: Could not capture frame
> [ERROR] [1589731616.195890605]: Could not capture frame
> [ERROR] [1589731616.229178206]: Could not capture frame

I've attached a rosbag of 60 msgs and the original mp4 here:
[small.mp4.zip](https://github.com/ros-perception/image_pipeline/files/4640567/small.mp4.zip)
[test.bag.zip](https://github.com/ros-perception/image_pipeline/files/4640568/test.bag.zip)

Running image_view on the rosbag's `/videofile/image_raw` as input gives an immediate seg fault.

> rosrun image_view image_view image:=/videofile/image_raw
> [ INFO] [1589732135.983422221]: Initializing nodelet with 4 worker threads.
> [ INFO] [1589732136.055495327]: Using transport "raw"
> Segmentation fault (core dumped)

aytimothy, 2020-05-18:
> @aytimothy Can you please test to see if the issue is resolved now that #526 is merged into Melodic?

I'll take a look see.

awesomebytes, 2020-05-18:
@hoonkai Thanks a lot for reporting and attaching the rosbag! I hope this is helpful for the developers :)
I'll look into the `Could not capture frame` error when I have some free time.

hoonkai, 2020-05-18:
@awesomebytes Thanks. Can you upload a working mp4 for me to test out `video_stream_opencv` + `image_view`? I've tried 5 different mp4s but none of them works.

hoonkai, 2020-05-18:
> @aytimothy Can you please test to see if the issue is resolved now that #526 is merged into Melodic?

That fixed it for me!

awesomebytes, 2020-05-18:
> @awesomebytes Thanks. Can you upload a working mp4 for me to test out `video_stream_opencv` + `image_view`? I've tried 5 different mp4s but none of them works and they all crash in the same fashion.

The mp4 found in video_stream_opencv was working I thought, up until your feedback. I can't take a look right now, but I'll do!

JWhitleyWork, 2020-05-18:
@hoonkai Please pull `master` and test. Once you've verified that the issue is resolved, I'll make a new release for Melodic.

JWhitleyWork, 2020-05-19:
> > @aytimothy Can you please test to see if the issue is resolved now that #526 is merged into Melodic?
> 
> That fixed it for me!

Awesome. I'll make release 1.15.0 for melodic now that includes this fix.

closed 2020-05-19 by SteveMacenski