ros-perception/image_pipeline #518: ROS2 image_proc decreases resolution far worse than melodic release

caelinsutch, 2020-04-16:
With the same config, the ROS2 version of image_proc has far worse resolution than the ROS1 version. Here's an example of the images:

https://imgur.com/a/g6EN5I0

SteveMacenski, 2020-04-16:
Do you see any clear errors or changes in the diffs between them?

caelinsutch, 2020-04-17:
Not that I can see when comparing melodic and ROS2 branch. I filed an issue in the opencv_vision package repo which provides the rectifiyImage method that's used in rectify.cpp

SteveMacenski, 2020-04-18:
Thanks for closing the loop on that one. If you find an issue and submit a PR, ping back here so I can be aware or push maintainers to react.

caelinsutch, 2020-04-22:
So the interpolation parameter doesn't seem to be working in the ROS2 release. The melodic release had the default interpolation mode to be `INTER_LINEAR`, versus ROS2 had it set default to `INTER_NEAREST`

 I'm not sure if this line: `interpolation = this->declare_parameter("interpolation", 0);` is working, as setting parameters in launch files and from command line doesn't seem to be working, so I've moved it into the parameter callback function for now (like the `image_mono` and `image_color` params 

@SteveMacenski what do you think about making a PR with those changes?

SteveMacenski, 2020-04-22:
It looks like the default in ROS1 is `1` or linear, so yes please, I'd accept a PR changing it back to that (https://github.com/ros-perception/image_pipeline/blob/melodic/image_proc/cfg/Rectify.cfg#L15).

On the question of the parameter not working - are you sure you have the namespacing correct? I'd do some parameter list and dumps to make sure its where you expect it to be.

caelinsutch, 2020-04-23:
As far as the parameter goes, based on the parameter lists and dumps it is in the correct place. After changing the interpolation param to be in the parameter callback function it worked fine to be set using parameters. Namespaces don't normally affect parameters used in launch files right?

closed 2020-04-23 by SteveMacenski