ros-perception/image_pipeline #525: Melodic on odroid-n2/ubuntu 18.04 cannot find ImageViewConfig.h

bltDefender, 2020-04-29:
`30%] Building CXX object abot-kinetic/image_pipeline/image_view/CMakeFiles/image_view.dir/src/nodelets/image_nodelet.cpp.o

Scanning dependencies of target image_rotate_gencfg
/root/catkin_ws/src/abot-kinetic/image_pipeline/image_view/src/nodelets/image_nodelet.cpp:34:10: fatal error: image_view/ImageViewConfig.h: No such file or directory
 #include <image_view/ImageViewConfig.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
abot-kinetic/image_pipeline/image_view/CMakeFiles/image_view.dir/build.make:62: recipe for target 'abot-kinetic/image_pipeline/image_view/CMakeFiles/image_view.dir/src/nodelets/image_nodelet.cpp.o' failed
make[2]: *** [abot-kinetic/image_pipeline/image_view/CMakeFiles/image_view.dir/src/nodelets/image_nodelet.cpp.o] Error 1
CMakeFiles/Makefile2:3205: recipe for target 'abot-kinetic/image_pipeline/image_view/CMakeFiles/image_view.dir/all' failed
make[1]: *** [abot-kinetic/image_pipeline/image_view/CMakeFiles/image_view.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....`

Can you please give me a hint what am I missing here? Where do I get the missing header file from?

JWhitleyWork, 2020-04-29:
@bltDefender This header file is generated by `dynamic_reconfigure` from the `image_view/cfg/ImageView.cfg` file when the node is built. Make sure you have installed all prerequisites using `rosdep` before trying to build the node and `catkin clean` your workspace and try again.

bltDefender, 2020-04-30:
I did already. All deps satisfied, ws was cleaned. From the build log I cannot see that `dynamic_reconfigure` writes that header file while others are written.

JWhitleyWork, 2020-04-30:
Can you give me more context? What command are you using to build the workspace? How is your workspace laid out? What version or commit hash of the repository and from which branch are you using?

bltDefender, 2020-05-01:
To ease the debugging I created a new workspace. In src I cloned recent branch `origin/melodic`. Above src folder I do resolve deps via `rosdep install --from-paths src -y --ignore-src --rosdistro=melodic`. Then I simply run `catkin_make`which fails . See https://pastebin.fun/6hamphEvyb

Ros Version is 1.14.5. Ubuntu ist bionic 18.04.4. Ros packages come from http://packages.ros.org/ros/ubuntu bionic main

bltDefender, 2020-05-03:
Is there anything wrong the way I did it?

JWhitleyWork, 2020-05-03:
@bltDefender If you are using packages from the apt repository then you would not have this issue because they would already be pre-built binaries. You are building a workspace which includes the source code of this repository. I need to know which version (tag or commit hash) of the source code you are building.

mikeferguson, 2024-01-17:
18.04 is now EOL

closed 2024-01-17 by mikeferguson